nightscout/AndroidAPS #1197: Wear OS: Complication tapping doesnt work on Wear OS 3.0 (Samsung Galaxy Watch 4)

ninelore, 2022-01-18:
Title

Expected Behavior: Open the Menu
Encountered Behavior: Nothing

Version: FB9325384-2022.01.15 (2.8.2.1)

dlvoy, 2022-01-18:
Tested and confirmed issue, working on fix.

Andries-Smit, 2022-01-18:
2022-01-18 21:59:00.029 28771-28771/info.nightscout.androidaps I/ComplicationTapBroadcastReceiver: onReceive: 
2022-01-18 21:59:00.049 28771-28771/info.nightscout.androidaps E/AndroidRuntime: FATAL EXCEPTION: main
    Process: info.nightscout.androidaps, PID: 28771
    java.lang.RuntimeException: Unable to start receiver info.nightscout.androidaps.complications.ComplicationTapBroadcastReceiver: java.lang.IllegalArgumentException: No injector factory bound for Class<info.nightscout.androidaps.complications.ComplicationTapBroadcastReceiver>
        at android.app.ActivityThread.handleReceiver(ActivityThread.java:3389)
        at android.app.ActivityThread.access$1200(ActivityThread.java:200)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1662)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6680)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
     Caused by: java.lang.IllegalArgumentException: No injector factory bound for Class<info.nightscout.androidaps.complications.ComplicationTapBroadcastReceiver>
        at dagger.android.DispatchingAndroidInjector.inject(DispatchingAndroidInjector.java:136)
        at dagger.android.AndroidInjection.inject(AndroidInjection.java:181)
        at dagger.android.AndroidInjection.inject(AndroidInjection.java:152)
        at dagger.android.DaggerBroadcastReceiver.onReceive(DaggerBroadcastReceiver.java:47)
        at info.nightscout.androidaps.complications.ComplicationTapBroadcastReceiver.onReceive(ComplicationTapBroadcastReceiver.java:51)
        at android.app.ActivityThread.handleReceiver(ActivityThread.java:3380)
        at android.app.ActivityThread.access$1200(ActivityThread.java:200) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1662) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.app.ActivityThread.main(ActivityThread.java:6680) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

PR #1198 (2022-01-18): Wear: fix complication tab dagger
Fixes #1197