nightscout/AndroidAPS #1199: RC3 / BYODA calibration / red triangle still there

robertrub, 2022-01-19:
@MilosKozak You had modified rhe code for this problem but it is still there. I'm calibrating G6/BYODA 4 minutes after last reading and I get a double in the BGs and rhe red triangle. 
The good part in RC3 is that the look stays closed.
[image]
[image]

MilosKozak, 2022-01-19:
i fixed it for G5. I'll check your G6 logs if it's the same

MilosKozak, 2022-01-20:
07:30:46.643 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.dexcom.cgm.EXTERNAL_BROADCAST Bundle{ sensorInsertionTime => 1641909055; senorInsertionTime => 1641909055; meters => Bundle[mParcelledData.dataSize=2764]; glucoseValues => Bundle[mParcelledData.dataSize=44124]; packageName => com.dexcom.g6.region3.mgdl; sensorType => G6; }Bundle
07:30:47.323 [pool-15-thread-3] D/DATABASE: [DexcomPlugin$DexcomWorker.doWork():185]: Inserted calibration TherapyEvent(id=1484, version=0, dateCreated=1642573847284, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1642573844000, utcOffset=3600000, duration=0, type=FINGER_STICK_BG_VALUE, note=null, enteredBy=null, glucose=104.0, glucoseType=null, glucoseUnit=MGDL)
07:30:47.334 [RxCachedThreadScheduler-86] D/CORE: [UserEntryLogger$log$2.invoke():45]: USER ENTRY: CALIBRATION Dexcom  [Timestamp(value=16425738 44000), TherapyEventType(value=FINGER_STICK_BG_VALUE), Mgdl(value=104.0)]
07:30:53.096 [main] D/DATABASE: [DataReceiver.onReceive():29]: onReceive com.dexcom.cgm.EXTERNAL_BROADCAST Bundle{ sensorInsertionTime => 1641909055; senorInsertionTime => 1641909055; meters => Bundle[mParcelledData.dataSize=2764]; glucoseValues => Bundle[mParcelledData.dataSize=44428]; packageName => com.dexcom.g6.region3.mgdl; sensorType => G6; }Bundle
07:30:53.590 [RxCachedThreadScheduler-59] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():35]: Firing EventNewBg GlucoseValue(id=31423, version=0, dateCreated=1642573853572, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573851000, utcOffset=3600000, raw=null, value=107.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
07:30:53.591 [RxCachedThreadScheduler-59] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():40]: Firing EventNewHistoryData GlucoseValue(id=31423, version=0, dateCreated=1642573853572, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573851000, utcOffset=3600000, raw=null, value=107.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
07:30:53.592 [pool-15-thread-2] D/DATABASE: [DexcomPlugin$DexcomWorker.doWork():160]: Inserted bg GlucoseValue(id=31422, version=0, dateCreated=1642573853571, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573843000, utcOffset=3600000, raw=null, value=108.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
07:30:53.593 [pool-15-thread-2] D/DATABASE: [DexcomPlugin$DexcomWorker.doWork():160]: Inserted bg GlucoseValue(id=31423, version=0, dateCreated=1642573853572, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573851000, utcOffset=3600000, raw=null, value=107.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
07:30:53.593 [pool-15-thread-2] D/DATABASE: [DexcomPlugin$DexcomWorker.doWork():164]: Updated bg GlucoseValue(id=31419, version=2, dateCreated=1642573853564, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=61e7af0a6cb3b9000415cd9c, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573542000, utcOffset=3600000, raw=null, value=109.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
GlucoseValue(id=31423, version=0, dateCreated=1642573853572, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1642573851000, utcOffset=3600000, raw=null, value=107.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
GlucoseValue(id=31422, version=0, dateCreated=1642573853571, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1642573843000, utcOffset=3600000, raw=null, value=108.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
07:30:55.644 [RxCachedThreadScheduler-56] D/NSCLIENT: [NSClientPlugin.onStart$lambda-5():124]: DBUPDATE entries Sent PairGlucoseValue 61e7af0a6cb3b9000415cd9c {"device":"AndroidAPS-DexcomG6","date":1642573542000,"dateString":"2022-01-19T06:25:42.000Z","isValid":true,"sgv":109,"direction":"Flat","type":"sgv"}31420/31423
07:30:57.664 [RxCachedThreadScheduler-56] D/NSCLIENT: [NSClientPlugin.onStart$lambda-5():124]: DBADD entries Sent PairGlucoseValue {"device":"AndroidAPS-DexcomG6","date":1642573843000,"dateString":"2022-01-19T06:30:43.000Z","isValid":true,"sgv":108,"direction":"Flat","type":"sgv"} 31421/31423
07:30:58.832 [RxCachedThreadScheduler-59] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():35]: Firing EventNewBg GlucoseValue(id=31422, version=1, dateCreated=1642573858821, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=61e7b0216cb3b9000415cda2, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573843000, utcOffset=3600000, raw=null, value=108.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
[... cut ...] =3600000, raw=null, value=107.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)
GlucoseValue(id=31422, version=1, dateCreated=1642573858821, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=61e7b0216cb3b9000415cda2, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1642573843000, utcOffset=3600000, raw=null, value=108.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_G6_NATIVE)

MilosKozak, 2022-01-20:
i'd need one more example. It's not so straightforward like G5

robertrub, 2022-01-20:
OK, I'll need to change my sensor tomorrow. I'll try to calibrate >4 min and see if triggers. Will post logs here.

robertrub, 2022-01-23:
@MilosKozak I tested at 3 min and 4 min after last BG but didn't get a double BG... Next time, I'll test just when 4 becomes 5 and see if I get the red triangle.
Will give more info here.

robertrub, 2022-01-24:
@MilosKozak 

It happened again. I think that I was atound 4.5 min past last BG.

[image]

Strangely, the app doesn't let me add the log file. I'll add via the web interface...

MilosKozak, 2022-01-24:
1st try https://github.com/nightscout/AndroidAPS/commit/1aa4f7452c58630c66dd4b56e6b4967341557700

MilosKozak, 2022-01-25:
disabled BOYDA calibration button
https://github.com/nightscout/AndroidAPS/commit/74daf2e7845c2568f1329208bc628eae2b5e8b6a

robertrub, 2022-01-25:
@MilosKozak Sorry but your _first try_ didn't solve the problem.

Entered calibration directly in BYODA maybe 15-20 seconds before BYODA reads the BG from the transmitter. I got the double BGs in AAPS. 
[image]
[image]

FYI, the ligne highlighted in RED is the "normal" reading and the one just above is the calibration value.

If it helps, the real timing was calibration followed by a "normal" reading but in AAPS, the order is reversed. "Normal" reading was registered first and the calibration BG was noted later.

Look at the logs around 19:29 to 19:31 (will add logs soon)

MilosKozak, 2022-01-27:
2nd  try https://github.com/nightscout/AndroidAPS/commit/900c5db0dc71c63ddef5e93586fbcd9b58588375

robertrub, 2022-01-28:
@MilosKozak  Seems like the 2nd try is the good one :)
Just tested at 12 sec before new BG read and didn't have a double in AAPS 
As you can see, the 92 (which is the "real blood" test result) arrived and stayed at +6 minutes and (I suppose) the BG via the transmitter at +5 min was deleted :)
[image]
Great job, tnx.
Leave this issue open for a couple of days so that I can test some more and add if needed.

closed 2022-01-29 by MilosKozak

vanelsberg, 2022-01-29:
Did not see this problem, tested anyway.
Did 6 calibrations using BYODA v1.9 (not started from AAPS) within 4-5 min: no "red triangle".
Tested OK!