nightscout/AndroidAPS #1201: Wear Tab Settings Access

bobhome007, 2022-01-19:
Reporting bugs
--------------
- Permanently an issue since RC1
- 09386a0b7-2022.01.18-22:21

Does not open wear settings since 3RC1 despite wear os being enabled and device working.

[image]

dlvoy, 2022-01-20:
Hi - I've retested and it works fine (3.0-rc3, Galaxy Watch 3)

Can you check if your `wear` and `app` apks are both build from exact same project (branch) and signed with exact same signing keys (JKS)?

If you have slightly older wear app than phone app it wont work. Or installed via pressing play button in Studio with default configuration, as it use debug keystore instead of personalized "production" one each user ought to prepare for themselves and use.
Can you see if commands from watch (like bolus configured on watch AAPS) gets approved on phone?

closed 2022-01-21 by MilosKozak

bobhome007, 2022-01-21:
yes bolus' do get approved by phone. Wear OS build works perfectly from RC3.
I built as release with a local config with identical signed details as a AAPS phone app.
 Just the AAPS settings tab on the phone does nothing which it did previously.