nightscout/AndroidAPS #1206: Feature Request: Autosense reset after changing faulty catheter / POD

filipwoz, 2022-01-20:
By using Omnipod it happens from time to time that on the last day the catheter is not delivering 100% of the ordered insulin (see also the expecience reported in https://de.loopercommunity.org/t/pro-und-contra-aaps-mit-omnipod-dash/8789) which causes the Autosense to go up in that time. After changing the Pod I expect that everything works again correctly - i.e. Autosense should be reset to initial state.
Currently it does not happen (tested with v3 RC1) and I needed to reset the Autosense using a workaround (changed the Autosense algorithm to another one and back to original one) to achieve it.
I thnik this would be usefull not only for Omnipod but generally by any fault catheter switch.

nicoleloop, 2022-01-21:
Oh my, could that be the reason why I run so low for the 12 hours following a Dash Pod change? I did not even think to check autosense, thank you for mentioning this issue, I'll use your work-around immediately as I just switched Pods myself.

avereha, 2022-01-21:
Please add more details:
 - what Autosens algorithm are you using?
 - screenshot with the "Autosens data"
 - screenshot with the Autosens output that does not reset after pod change.

avereha, 2022-01-23:
Any screenshot that shows when Autosense is not reset after pod change.

Freloner, 2022-01-28:
Hey there, just changed a pod and Autosens is reset !

[Logs link, around 08:33](https://linksharing.samsungcloud.com/iO2bennZulGl)

[image]

[image]

MilosKozak, 2022-01-29:
@filipwoz are you able to replicate?

nicoleloop, 2022-01-29:
Yes I was able to get proof of autosens Not Resetting after a pod change.

filipwoz, 2022-01-29:
@MilosKozak I will let you know once I change the pod once again. By the last 2 ones I had some other problems and I was not able to observe it properly.
I have however one more idea what might be "wrong" here. I observed that also by profile change Autosense is not being reset immediately - it always takes some minute or two until I can see again 100% by the Autosense Icon. So maybe it is not that it is **not reset** but the reset is only **delayed**?
I will observe it next time and come back to you.

MilosKozak, 2022-01-29:
you can observe it first after next BG

filipwoz, 2022-01-31:
OK, yesterday I observed  the Autosense after the POD change and it got reset after the next BG. So looks like by my original observation I had too less patience.

closed 2022-01-31 by filipwoz