nightscout/AndroidAPS #1208: Feature request: Automating backups/export settings

ninelore, 2022-01-20:
Maybe add something to automathe the creation of settings exports / backups.

vanelsberg, 2022-01-20:
Yes. Good idea.
I would definitely vote for such functionality:

- Especially useful for OmniPod EROS/DASH users as Pod state is part of settings:
To reinstall or in case of problems, restoring settings with an active Pod is only possible with settings exported for the active Pod.
- Settings contain users Objectives. Also a reason to make regular backups of settings to make sure they are not lost.
- Settings contain current profile settings. They are also to valuable to not backup.

vanelsberg, 2022-01-20:
I can think of a number of solutions:
- **EROS/DASH**
    - Automatically export after Pod activation

- **General**:
    - Automatically backup settings every NN days, Optionally NN being a user configurable setting?
    - Automatically backup settings on profile change?
    - Add it as an action under automation?

robertrub, 2022-01-21:
It would be nice to have the file written or copied to a cloud drive too.

Freloner, 2022-01-21:
Yes all your idea are nice ! 🥂

andyrozman, 2022-01-21:
@robertrub On old repo we had ticket opened, so that backups could be made to Dropbox, but it was never implemented... Maybe we need to add it here again as new feature request... It is something I would also like to have... But I imagine if we open it this FR will be done after 3.0 release...

Philoul, 2022-01-23:
Dropbox interface is probably too complicated for me, but I can try to propose something... (new automation action to export preferences, all existing automation triggers available, I just have to check how to manage password...)

vanelsberg, 2022-01-23:
> Dropbox interface is probably too complicated for me
And then there is OneDrive and Google drive. To complicated, yes.

How about an option to export settings and then copy it to an alternate (user configurable) location. Automation could handle that. User then needs to choose a directory that is synced by the phone to whatever cloud storage it is configured for.

ninelore, 2022-01-23:
i was mainly thinking about getting an export automation actions and let the upload to wherever be handled by the apps of these cloud platforms

lee-b, 2022-04-04:
Automated export to a selected folder would be fine.  Then people can use any drive/sync/backup tool they like.

erobinson, 2022-04-25:
Not fully automated, but made is easier to upload/email exported settings - https://github.com/nightscout/AndroidAPS/pull/1674

ffsb42, 2023-04-01:
+1 to all options above:

1. automatically save settings on a regular basis (every day, keep last 5 days, last 4 nondays, last 12 months)
2. automatically save settings after enabling a new omnipod (or other event requiring data from settings)
3. provide an option to keep a copy of the settings in an alternate location
4. provide an option to keep a copy of the settings in a predefined cloud location (note IMHO, since we are on android, google drive might be better suited than dropbox)

rudiru, 2023-12-16:
The password should be stored so that it is only needed to restore a backup and not to create one.

FalconEagleTest, 2024-06-16:
hi,
any progress with the auto cloud backup?
it will make restore to functionality very fast in case of broken/lost phone,
thanks.

olorinmaia, 2026-09-23:
https://github.com/nightscout/AndroidAPS/pull/4481 & https://github.com/nightscout/AndroidAPS/pull/3487 is merged. Closing issue

closed 2026-09-23 by olorinmaia