nightscout/AndroidAPS #1217: 3.0rc1 - Omnipod Dash - Cannot "Discard Pod" after fault when setting up.

StephenBrown2, 2022-01-24:
While attempting to start a new Dash Pod, it faulted with code 016 ALARM_SAWCOP_RESET. I then had to disable the screaming pod by breaking the board as instructed here https://loopkit.github.io/loopdocs/faqs/omnipod-faqs/#what-do-you-do-to-stop-a-screaming-pod

I however could not get a new Dash pod to connect, as it still attempted to connect to the old one.

Things I have tried, unsuccessfully, to clear the stuck dash pod status:

- I can switch back and forth between Eros and Dash, and still the fault remains in AndroidAPS for the Dash.
- I have cleared the Bluetooth cache
- I have cleared the app cache
- I have reset the AndroidAPS databases
- I have exported the settings, cleared the AndroidAPS app storage, and re-imported the settings
- I have upgraded to dev, which includes https://github.com/nightscout/AndroidAPS/pull/1212

I am now running with an Eros pod, which connected via my OrangeLink successfully and has had no issues, but still the dash pod fault remains in AndroidAPS.

Potential solution:

I would like the "Discard Pod" button to become available when during setup, so that a faulted pod does not prevent any new pods from being connected and we can clear any stored information about a Dash pod that is no longer reachable/usable.

Possibly related (In that the proposed solution might help in those situations):
- https://github.com/nightscout/AndroidAPS/issues/1159
- https://github.com/nightscout/AndroidAPS/issues/1016
- https://github.com/nightscout/AndroidAPS/issues/935

Also possibly related: https://github.com/nightscout/AndroidAPS/issues/1208

I could not get it to "forget" the last pod state, because it was stored with the settings that were exported. As soon as I re-imported my settings, the same Pod Fault message would come back.

Screenshots:

Omnipod Dash config screen
[image]

Omnipod Dash Pod Management screen:
[image]

Unfortunately I did not get a screenshot of the Dash Pod History screen.

Attempting to refresh to clear the error doesn't work:
[image]

Omnipod Eros Pod Management screen:
[image]

Logs:
AndroidAPS 3.0-rc1
Build: 86ec769-2022.01.12-18:28
Remote: git@github.com:nightscout/AndroidAPS.git
Flavor: fullrelease
[AndroidAPS_LOG_1642977024009.log.zip](https://github.com/nightscout/AndroidAPS/files/7922278/AndroidAPS_LOG_1642977024009.log.zip)

I'm not sure of the exact time of the issue, but it started at some time between 18:25 and 18:45 on 2022-01-23, and I believe I finally gave up and switched to an Eros pod between 18:45 and 19:00

vanelsberg, 2022-01-24:
The "Deactivate" button is disabled for DASH while showing "setup in progress"?

vanelsberg, 2022-01-24:
As a work around, If you have a settings file that was exported for a pod previous to the current one (that faulted) you will import an "old" pod. You can then hit "Deactivate". It will obviously fail with an error (as the Pod does not exist).
Het retry until you get the option to "Discard".

From there you can activate a new Pod.

avereha, 2022-01-24:
You should use the 'Deactivate pod" in this case.

StephenBrown2, 2022-01-25:
Thanks, I'll give that a try when my current pod expires tomorrow, but it would be a lot easier if the "Discard" button showed up when there is a Fault received from the Pod so one doesn't have to know to attempt to "Deactivate" multiple times.

StephenBrown2, 2022-01-27:
This turned out to be true. The Discard Pod option showed up in the Deactivate Pod window after trying to discard the now absent pod.

I still think this would be better served by a Discard Pod button matching the layout in the Eros Pod Management window.

avereha, 2022-01-27:
Most of the time you can deactivate a pod with a fault

MilosKozak, 2026-10-13:
In AAPS 4 (`dev`, from 3a103c4c27) a pod that faults part way through activation can be removed from the Omnipod screen:

- "Deactivate pod" is shown as soon as the pod is paired, not only after activation is complete. This is for Dash and Eros.
- Most pods with a fault can still be deactivated. If deactivation fails, the deactivation wizard offers "Discard pod".

So the old settings file trick is not needed any more.

@StephenBrown2, can you please re-test with v4 when you can, and tell us if this works for you?

closed 2026-10-13 by MilosKozak