nightscout/AndroidAPS #1218: [v3-RC3] NO APS SELECTED OR PROVIDED RESULT

Andries-Smit, 2022-01-24:
When upgrading to v3, On the loop tab there was a message "NO APS SELECTED OR PROVIDED RESULT" With my omnipod eros and Oref1. 
After disabling the autosens, it started to work.
When the pump did run for 30 minutes, the autosens could be switched on again.

This was cause by the database being empty and not containing enough BG history to calculate autosens

Suggestions: 
- document to switch it off when upgrading. (However, users might forget to switch it on again)
- Or fix it, let the loop work with autosens suppressed when there is not enough data, and show a notification for the user. And start working with autosens when there is enough data

MilosKozak, 2022-01-25:
from what i see the reason is mentioned on APS tab

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak

oakeley, 2023-11-11:
How can I turn autosens on/off
This bug basically prevents AAPS from working as it does nothing in this state. Just acts as a manual control for my pump using my phone.
Please can someone remind me where in the hundreds of config settings or preference settings I can find a tab to turn autosens on/off for 30 minutes? Perhaps a button on the loop tab would be helpful. My AAPS has been down for about 10 days so far. Which is quite stressful. Many thanks for your help.