nightscout/AndroidAPS #1219: Eat Reminder (high glucose/time)

annushkiii, 2022-01-24:
Hello, not issue, but improvement idea.

when using the bolus wizard my sugar is over my target, I get a pop up like "you shoud eat later [...] should remind you?". If I choose yes, the reminder rings only when my sugar is unter/at my target. 

This is confusing, because I can enter a reminder time. I assume that I get a reminder exact in the time, which I enter. 

If I enter for example 30 minutes, and my sugar is over my target to get a reminder "You should eat, but your sugar is over target. Should remind when you reached your target?" or somehing like this.

MilosKozak, 2022-01-24:
haw can you enter reminder time?

annushkiii, 2022-01-24:
[image]

annushkiii, 2022-01-24:
[image]

closed 2022-01-24 by MilosKozak