nightscout/AndroidAPS #1223: Feature Request: Leak detection

crizz6396, 2022-01-26:
One of the biggest issues with pods is occlusions & leakage.  If there was a way to identify a leak by too little pressure that would be incredibly useful.

MilosKozak, 2022-01-26:
I believe it's close to impossible distinguish between sensitivity change, forgotten bolus, leaking infusion set etc within reasonable period of time

closed 2022-01-26 by MilosKozak

crizz6396, 2022-01-26:
It would have to be during a bolus.  Even if there was a way to show the
pressure change.  I'm assuming an occlusion is detected when pressure
builds up.  What I don't know is how omnipod or other pumps report the
value.  It would just be a useful tool to have when troubleshooting.

Yesterday I went live for the first time and trying to determine why all of
a sudden my bg was going high after bolusing multiple times.  It appeared I
wasn't getting all my insulin.  It was almost like I was getting no basal.
Hopefully today goes better.

MilosKozak, 2022-01-26:
this value is not sent outside the pump

crizz6396, 2022-01-26:
Damn...