nightscout/AndroidAPS #1227: 3.0 RC3 | No bg received while values in xdrip tab

bargro1991, 2022-01-26:
This is the second time I experience this issue, using FSL2/xDrip+/ and AAPS 3.0 Rc3.
AAPS is stating the last received BG is 149 minutes ago, while in de xDrip tab it shows the BG values like it should.

[image]
[image]

MilosKozak, 2022-01-26:
log file?

bargro1991, 2022-01-26:
Apologies
[AndroidAPS_LOG_1643205481190.log.zip](https://github.com/nightscout/AndroidAPS/files/7944480/AndroidAPS_LOG_1643205481190.log.zip)

bargro1991, 2022-01-27:
i'm afraid the logs that correspond with these events are gone, is there any way to get them back?

MilosKozak, 2022-05-27:
should be fixed in dev

closed 2022-05-27 by MilosKozak