nightscout/AndroidAPS #1230: No glucose register when use calc

mick711, 2022-01-26:
It's happen with Dev 8fb17, when use Calc no glucose register but when use glucose it's OK. 
Phone zenfone 8, android 12.

[image]
[image]
[image]
[image]
[image]

Log file: https://drive.google.com/file/d/1p7FL8g-ijHBEXwn7wxLevvb9LKTtjurn/view?usp=drivesdk

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-01-26:
log is empty. drag&drop the file from phone here

mick711, 2022-01-27:
[AndroidAPS_LOG_1643196772856.log.zip](https://github.com/nightscout/AndroidAPS/files/7948175/AndroidAPS_LOG_1643196772856.log.zip)

I can make a new log file for lunch .

mick711, 2022-01-27:
I think issue can be closed. With a20f4f.. Dev built this issue not happen

closed 2022-01-27 by mick711