nightscout/AndroidAPS #1234: 3.0rc3/Combo - Incorrect IOB after Occlusion alarm during prime

T-o-b-i-a-s, 2022-01-27:
Background
--------------
- Had an occlusion at night at around 2:40 and thought it was in the catheter, which I replaced
- When priming, another occlusion alarm was raised as the occlusion seemed to be in the tube.
- The pump and the phone were ringing like crazy, to not wake up my whole family I immediately shut down all alarms without reading any detailed error messages in AAPS
- After exchanging the tube and priming, I was surprised to have IOB above 2 in the middle of the night
- It turned out that AAPS did not mark the interrupted bolus as prime, but as a regular bolus
- When deleting the bolus in the Treatment list, the IOB was reasonable.
- In xDrip, I later saw the error message from AAPS to check Insulin Treatments

Version
-----------
- AndroidAPS 3.0rc3
- Phone: Oneplus Nord with LineageOS 17.1 (Android 10)
- Dexcom BYODA 1.7.0.3
- Combo Pump

Suggestions
------------------
- When Prime is interrupted, record the time and amount the interruption occurred and sync the info with the treatment history the next time the pump history is read
Or
- Use a permanent Error message type (like the one informing about new AAPS versions) to warn the user that the IOB might be incorrect to make sure the message can be read without being stressed by the ringing alarms

If you need the log files, I am happy to provide them.

MilosKozak, 2022-01-27:
Not every situation can be resolved in AAPS expecially with interrupted comm, failed bolus etc. @jotomo ?
Rule No.1 is when anything goes wrong always check pump and AAPS state and do manual correction (exactly what you did)

jotomo, 2022-01-27:
What @MilosKozak said. GIven the limitations of the Combo comms, there are edge cases the user needs to check and correct if needed like you did. The popup message you saw just stated generically that something went wrong, asking the user to check. Turning that into a persistent notification would be redundant. While a technical solution to marking an interrupted bolus as prime, like proposed, would be possible, this would only cover one of many edge cases involving pump errors and implementing this would likely introduce multiple new bugs as complexity increases considerably.

T-o-b-i-a-s, 2022-01-28:
Thanks for the fast and detailed feedback. You are correct that covering this edge case is probably too much effort for the probability it can occur. It was the first time such a situation happens to me during my 4 years if AAPS usage.
I will close the issue, thanks for your great work and engagement!

closed 2022-01-28 by T-o-b-i-a-s