nightscout/AndroidAPS #1239: [v3-rc3] Bolus Dialog does not close on cancel

Andries-Smit, 2022-01-28:
When cancelling a bolus in the progress with an Omnipod Eros, with the cancel button, the dialog shows the message below. The pumps cancels correctly, and the dialog can not be closed. However, before version 3 the dialog was removed/closed when the cancel was completed. 

Workaround, minimize app, and reopen.

[image]

It seems that the timestamps do not match
https://github.com/nightscout/AndroidAPS/blob/dev/core/src/main/java/info/nightscout/androidaps/dialogs/BolusProgressDialog.kt#L134

13:35:18.414 [main] D/PUMP: [BolusProgressDialog.onResume$lambda-3():133]: Running timestamp 1643373302482. Close request timestamp  1643373307351
[AndroidAPS._2022-01-28_00-00-01_._cancel_issue.zip](https://github.com/nightscout/AndroidAPS/files/7959560/AndroidAPS._2022-01-28_00-00-01_._cancel_issue.zip)

MilosKozak, 2022-01-29:
The timestamp is used to distinguish beteween bolus and SMBs. Can you debug it? Driver should use EventOverviewBolusProgress.Treatment

vanelsberg, 2022-01-31:
Confirming: also seeing this.

regiorunner, 2022-02-01:
This issue seems related to #987 [3.0-beta6] Eros: Message bolus stopped not cleared.

closed 2022-06-21 by MilosKozak