nightscout/AndroidAPS #1241: 3.0 RC3 Unable to activate profile

HuismanErik, 2022-01-29:
Hey guys,

Today I tried to adjust something in my profile. Unfortunately, that didn't work out. And when activating AndroidAPS did a profile switch.
And the pulldown didn't let me select the right profile anymore. That's why I deleted the unwanted profiles. In the end, it seems to me that it is no longer necessary to activate the profile. When I press the activate profile button nothing happens. Below I have posted a number of screenshots and posted logging. I hope you can help me!

[image]
[image]
[image]

[AndroidAPS_LOG_1643463038940.log.zip](https://github.com/nightscout/AndroidAPS/files/7963798/AndroidAPS_LOG_1643463038940.log.zip)

Sincerely,

Erik

HuismanErik, 2022-01-29:
Okay, I managed to reactivate the profile by making a backup and then restoring a backup from a week ago. Then reload the latest backup. I'm glad it's working again, just don't dare to adjust the profile anymore.

MilosKozak, 2022-01-29:
if dialog for profile doesn't appear no valid profile is detected. If you cannot find the bug in profile you can always remove it and create new

closed 2022-01-29 by MilosKozak

HuismanErik, 2022-01-29:
A new profile could also not be activated with the activate button. I have tried everything including a new profile and finally a database reset. After restoring a backup, the profile was active again. I was pretty desperate.
It is not clear to me what is wrong with the profiles I create. I can't find anything in the log file about the profile being disabled.
In the screen the profile seems to be good, there is no red background. The profile has also worked well for at least a month.

MilosKozak, 2022-01-30:
you can sent me non working configuration for more analysis

HuismanErik, 2022-01-30:
I have looked at the code and my problem has been solved by making a small change to the code. See the PR. Anyway, thanks for your offer.