nightscout/AndroidAPS #1242: [3.0-rc3] no button for catheter/reservoir/pod change

annushkiii, 2022-01-29:
Hello,

everytime I change the pod, I give 1.8 iE for the change. In the settings I choose a Fill/Prime standard insulin amount of 1.8. iE. In my action tab I can't choose something like catheter/reservoir/-exchange (screens below). 
[image]
[image]

Philoul, 2022-01-29:
I think prime/fill is automatically done when you change your pod (no risk of error because build in values).
These settings are required for other pumps because tubing and catheter are not all the same.

MilosKozak, 2022-01-30:
the button is not there because omnipod doesn't support priming at all. having it there could be very confusing for most people

annushkiii, 2022-01-30:
@MilosKozak @Philoul 

A part of Omnipod users giving after podchange an extra basal to fill the little needle with insulin. Not all needed this, but in my opinion a lot. I'm in Facebook in a german Omnipod Group and read very often, that it is required.

My problem with AAPS because of the lack of  this button is, that I give the 1,8 needed bolus (this works fir me really good). The algorhitmus thinks "1,8 iE extra without carbs? Oh I have to reduce the basal!" and this is wrong. Because of this I asked in Looped DE for a solution and the recommendation is posting this as an issue here.

vanelsberg, 2022-01-31:
> Facebook in a german Omnipod Group and read very often, that it is required.
Never heard of this. AFAIK this is utterly non-sense,

Omnipod primes automatically after canula insert.
Check the Omnipod user manual. There is no mention on requirement nor recommendation to do any additional "priming".

annushkiii, 2022-01-31:
I knew the manual and use omnipod since over 5 years. Maybe I explain it here wrong. 

Started now a survey at Facebook and post the result here after few days.

annushkiii, 2022-02-02:
Now I post the results. There are very different. Sry it's in German but I can translate. I asked if someone use extra bolus after podchange. I asked this in Looped-DE group amd in Omnipoddies & Friends group. Looped Community don't need the extra bolus (I think, because loop can handle it self). The people from Omnipoddies group needed it. Someone post a workaround, which said to enter few carbs, this don't confuse the loop, I think it's a good solution.

Ja=yes
Nein=no

[image][image]

Andries-Smit, 2022-02-03:
@annushkiii a work around is; do add the insulin, like would do as a bolus. Thereafter you remove the in the treatment dialog (via de option menu, treatments), When remove it from the history and the algo will not use it the calculate the basal rate.

annushkiii, 2022-02-04:
> @annushkiii a work around is; do add the insulin, like would do as a bolus. Thereafter you remove the in the treatment dialog (via de option menu, treatments), When remove it from the history and the algo will not use it the calculate the basal rate. 
> 

Thx this is a good solution!

closed 2022-02-08 by MilosKozak

Andries-Smit, 2022-06-20:
@annushkiii A reference on the why you might need of an extra bolus: https://www.omnipod.com/poddertalk/diabetes-support/why-so-high-that-annoying-new-infusion-site-phenom