nightscout/AndroidAPS #1243: Feature request: Add new buttons for easier UAM looping

a-rigatuso, 2022-01-29:
When using no-carb/bolus looping it would be helpful to have more options for buttons in overview settings. I would like to add **temporary target presets (eating soon, activity, hypo** from the temporary target dialogue) and **profile switch** (from action tab) as optional buttons on the home screen.

I built a quick mockup for clarification:  

[image]

With active eating soon TT:  

[image]

Philoul, 2022-01-29:
full UAM looping is not standard way to loop.
You can access to profile switch with loog click on profile name, and to all other functions from with just one more click on target button.
I'm afraid your proposal will incease the complexity in settings (several additional buttons).

MilosKozak, 2022-01-30:
you can create custom buttons with 3.0 and automation
this is edge case to be implemented to main AAPS

closed 2022-01-30 by MilosKozak