nightscout/AndroidAPS #1244: Dev 20/ Crazy calculation "loop" on main UI with strange graphics

robertrub, 2022-01-30:
From time to time (didn't find what triggers it), there is a _recalculation new data_ that starts over and over for a long time with strange graphs in the middle of the calculations.

Eventually it stops recalculating qnf the graph goes back to normal. So, not a major bug but seems strange.

[image]

For the logs, look at around 8h54

MilosKozak, 2022-01-30:
can you paste screenshot of Treatments Temp basals?

robertrub, 2022-01-30:
No change in temporary basals. Temp target change but several minutes before the "loop".
Different captures around 8:54

[image]
[image]
[image]

The bolus was injected at 8:54 once the calculations were finished.
[image]

Thomas-Wolff, 2022-01-30:
I can also confirm this, but it settles down again after a few seconds. 

Have a nice Sunday, greetings Thomas

robertrub, 2022-01-30:
It took about 1 min in this case (usually, it takes less time).

vanelsberg, 2022-01-31:
Seeing this also when changing scale. But it takes only a few seconds (up to 10?) before it settles down.

closed 2022-03-07 by robertrub