nightscout/AndroidAPS #1257: [3.0.0] CGM Eversense integration does not work

BernhardRo, 2022-02-01:
AAPS Version: 3.0.0
Build Version: a2411bb51-2022-02-01-08:40
Flavour: fullrelease
NS Version: 14.2.5

Hello,
as already recognized for the RC3 version, the BG reading from Eversense does not work in AAPS 3.0.0. For older versions, it worked with patched ES apk -> ESEL -> MM640g in AAPS.
But this does not work with the new AAPS version. The BG values are simply not available. As workaround, it is possible to get them via xDrip+ (and 640g / Eversense as source in xDrip+), but the direct connection seems to be broken.

Regards
Bernhard

AdrianLxM, 2022-02-02:
@BernhardRo 
Could you please set a break point in line 93 of the `LocalBroadcaster` and check if you get the AAPS package name?
 
```
 ​            ​if​ (packageName ​!=​ ​null​) {
```

BernhardRo, 2022-02-02:
Yes, I have it:
[image]

MilosKozak, 2022-02-03:
put here logs from AAPS (be sure there were some BGs sent)

BernhardRo, 2022-02-03:
ADD TIME OF EVENT HERE:
ADD ISSUE DESCRIPTION OR GITHUB ISSUE REFERENCE NUMBER:
-------------------------------------------------------
(Please remember this will send only very recent logs.
If you want to provide logs for event older than a few hours,
you have to do it manually)
-------------------------------------------------------
AndroidAPS 3.0.0
Build: a2411bb51-2022.02.01-08:40
Remote: https://github.com/BernhardRo/AndroidAPS.git
[AndroidAPS_LOG_1643909349641.log.zip](https://github.com/nightscout/AndroidAPS/files/7996882/AndroidAPS_LOG_1643909349641.log.zip)

Flavor: fullrelease
Nightscout-Version: 14.2.5

BernhardRo, 2022-02-03:
I have a debug session with @AdrianLxM on Saturday