nightscout/AndroidAPS #1258: 3.0.0 NS Client no Control

elgrupo123, 2022-02-01:
NS Client 3.0.0 control.

With NSClient 3.0.0, Profiles/Temporary Rate on the Master cannot be controlled/changed. Even the last action such as changing the pump is not displayed. With RC3 it worked without problems. Also NS shows all NS client data. Bug or are we making a mistake??

reijoahola-nightscout, 2022-02-01:
It looks like s ome of NSClient synchronisation settings are removed in AAPS. Profile Switch and Temp Target for example. I do not know the reasons.

elgrupo123, 2022-02-01:
In the RC3 version, it worked wonderfully and made life a lot easier for us parents. If it is wanted now, it would be a huge step backwards and a shame.

reijoahola-nightscout, 2022-02-01:
Seems to be fixed now in 3.0.1.

elgrupo123, 2022-02-01:
3.0..0.1 does not remedy the situation. 3.0.1 is currently having problems downloading.

MilosKozak, 2022-02-01:
with 3.0.0.1 you can enable it in preferences

closed 2022-02-01 by MilosKozak