nightscout/AndroidAPS #1265: Temp zero basal after profile switch is not respected by pump (omnipod eros)

sennierer, 2022-02-03:
When switching to a profile (in my case one altered on NS) AAPS sets a 4800 minutes temp basal zero. However, in the omnipod tab under temp basal rate there is a dash only and I heard from the clicking that the pump is delivering insulin (while AAPS records zero).

Screenshots is attached (profile switch was at 7 in the morning).
Build is 23207a275-2022.02.01-20:24, Orangelink v1 firmware 3.2. I am using a Pixel 3a with Grapheneos latest release.

Relevant part of AAPS log is [here](https://gist.githubusercontent.com/sennierer/456955c098392995e7d15605337bf0f6/raw/0b2d24d31854178becf267dcd6509bfeffebd013/aaps_log_profile_switch_3-2-22.log)

[image]

sennierer, 2022-02-03:
Sorry, GitHub didnt take the second screenshot the first time ... 

[image]

philippowell2557, 2022-02-12:
I'm having this issue as well! Please share if you have found a resolution to the indefinite temp basal of 0 units / hour.

samspycher, 2022-05-23:
Hi @sennierer, as I read from the code, the 80 hours is a default duration for a "fake suspended TBR" that is created temporarily during certain workflows. It is cancelled when delivery continues, and in fact you do see the log entries "Cancelling fake suspended TBR" in your log file when normal delivery is resumed. 

Perhaps the issue is that the cancellation of the fake TBR is perhaps not as visible on the UI as the creation itself?

It would be helpful to see a larger scope of the log, at minimum including the time of the screenshots, and also screenshots of the treatment decisions. If this is something that is re-occurring, it would be great if you could send a more complete trace along with screenshots also of treatment decisions.

If it does not occur anymore, I suggest we close this issue.

nlof, 2023-04-17:
I seem to have the exact same problem with Dash. Temp basal of zero in manual mode gets overridden very frequently, it appears to have to do with the frequent profile switches.
[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11253233/AndroidAPS.log)
[AndroidAPS._2023-04-17_18-22-17_.3.zip](https://github.com/nightscout/AndroidAPS/files/11253237/AndroidAPS._2023-04-17_18-22-17_.3.zip)
[AndroidAPS._2023-04-17_18-22-17_.1.zip](https://github.com/nightscout/AndroidAPS/files/11253240/AndroidAPS._2023-04-17_18-22-17_.1.zip)
[AndroidAPS._2023-04-17_18-22-17_.0.zip](https://github.com/nightscout/AndroidAPS/files/11253241/AndroidAPS._2023-04-17_18-22-17_.0.zip)
[AndroidAPS._2023-04-17_00-00-00_.72.zip](https://github.com/nightscout/AndroidAPS/files/11253242/AndroidAPS._2023-04-17_00-00-00_.72.zip)
[AndroidAPS._2023-04-17_00-00-00_.71.zip](https://github.com/nightscout/AndroidAPS/files/11253243/AndroidAPS._2023-04-17_00-00-00_.71.zip)
[AndroidAPS._2023-04-17_00-00-00_.70.zip](https://github.com/nightscout/AndroidAPS/files/11253245/AndroidAPS._2023-04-17_00-00-00_.70.zip)
[AndroidAPS._2023-04-17_00-00-00_.69.zip](https://github.com/nightscout/AndroidAPS/files/11253248/AndroidAPS._2023-04-17_00-00-00_.69.zip)
[AndroidAPS._2023-04-17_00-00-00_.68.zip](https://github.com/nightscout/AndroidAPS/files/11253249/AndroidAPS._2023-04-17_00-00-00_.68.zip)
[AndroidAPS._2023-04-17_00-00-00_.67.zip](https://github.com/nightscout/AndroidAPS/files/11253252/AndroidAPS._2023-04-17_00-00-00_.67.zip)
[AndroidAPS._2023-04-17_00-00-00_.66.zip](https://github.com/nightscout/AndroidAPS/files/11253253/AndroidAPS._2023-04-17_00-00-00_.66.zip)
[AndroidAPS._2023-04-17_00-00-00_.65.zip](https://github.com/nightscout/AndroidAPS/files/11253255/AndroidAPS._2023-04-17_00-00-00_.65.zip)
[AndroidAPS._2023-04-17_00-00-00_.64.zip](https://github.com/nightscout/AndroidAPS/files/11253256/AndroidAPS._2023-04-17_00-00-00_.64.zip)
[AndroidAPS._2023-04-17_00-00-00_.63.zip](https://github.com/nightscout/AndroidAPS/files/11253259/AndroidAPS._2023-04-17_00-00-00_.63.zip)
[AndroidAPS._2023-04-17_00-00-00_.62.zip](https://github.com/nightscout/AndroidAPS/files/11253261/AndroidAPS._2023-04-17_00-00-00_.62.zip)
[AndroidAPS._2023-04-17_00-00-00_.61.zip](https://github.com/nightscout/AndroidAPS/files/11253263/AndroidAPS._2023-04-17_00-00-00_.61.zip)
[AndroidAPS._2023-04-17_18-22-17_.2.zip](https://github.com/nightscout/AndroidAPS/files/11253238/AndroidAPS._2023-04-17_18-22-17_.2.zip)

Attached logs, probably the most recent would have been enough...