nightscout/AndroidAPS #1270: Support for Multiple Active Quick Wizards

rustymonkey, 2022-02-03:
Currently only a single Quick Wizard is displayed for a given time period. This limits their usefulness since it means that users must switch between quick wizards and 'Carb' or 'Calculator' entries. e.g. they'll use:
- Quick wizard 'Lunch' for 40g lunch
- 'CALCULATOR' for 20g desert

For a user who wants to use 'Quick Wizards' it would be preferable that they could use quick wizards for all entries associated with a meal. e.g.
- Quick wizard 'Meal' for 40g lunch
- Quick wizard 'Snack' for 20g desert

A possible implementation would be that if multiple quick wizards are active at a given time:
- A 'Quick Wizard' button would display with the description 'Quick Wizard'
- Clicking the button would display a 'Quick Wizard' dialog showing all wizards active at that time. So the user could then click the relevant button.

Note:
- Using tap-hold to open the dialog containing all the active wizards isn't a preferable approach due to discoverability issues. 

Aside: 'Quick Wizard' isn't descriptive of the functionality. It would be more understandable if the name was changed to e.g. 'Meal Shortcuts'.

Andries-Smit, 2022-02-03:
@rustymonkey, i implemented this for the wear os tile... pull request will be available, after #1160 is merged.

Your addition too implement it for the phone app seems valid for me

MilosKozak, 2022-02-07:
quickwizard is designed to be single click functionality. this is adding more user interaction. i'm not sure about it but keeping it here as enhancement

Andries-Smit, 2022-02-07:
@MilosKozak can we make it a user choice by configuration? 
If users configure 2 in parallel, it is not perfect, it show only the first one as active. We can make it happen when multiple quick items are active, we show a dialog in between. Keeping the current function as is, and improving for the ones who implement multiple items

closed 2024-03-18 by MilosKozak