nightscout/AndroidAPS #1272: Add Motion Detection to Automation

bobhome007, 2022-02-03:
Enhancement
Add an automation event type called "motion detect"

The setting should use the Android API for the accelerometer on the phone or watch to detect, walking, standing, running and driving with another setting for level of intensity from 1 to 100.

Based on the triggered event the usual Automation actions should be available.

Eg.
Event:
Motion Detection = eRunning
Intensity = 40%

Action: set a profile % of 70% for 35minutes

bobhome007, 2022-02-03:
Enhancement

reijoahola-nightscout, 2022-02-04:
I have tested xDrip+ Motion detection feature with my Galaxy phones. Galaxy S9 could use it but Galaxy S21 G5 did not like it and xDrip+ always turns Motion Detection off after about 15 min. There is something incompatibility and differences between phone models even from the same manufacturer.

bobhome007, 2022-02-04:
If AAPS has the feature, it will be able to automate profile changes which xDrip cannot do.

Also if the feature is added to AAPS it can be enabled only for certain tested for phones. This means if the phone does not pass a pre validation teat "what hardware are you" then AAPS will not enable the feature.

Andries-Smit, 2022-02-04:
An interesting feature for automation
The Activity detection API  https://developers.google.com/location-context/activity-recognition give some possibility, before this API it was just too energy-consuming. But still, we should keep our battery consumption as low as possible; "No battery = no pump adjustments"

bobhome007, 2022-02-04:
Hi yes this is the part of the Api I was referring.  I was aware of it which better supports movement intensity and takes advantage of the advancement in cpu cores and low power consumption.

I think it would be a nice feature.

MilosKozak, 2022-02-07:
If I remember well, accessing health data was limited to play store applications. @AdrianLxM ?

JazzZ1, 2022-02-25:
Would this be possible with heart rate monitor like with Samsung Health by Galaxy Watch too? That would be great, having profile changed by like HR 140 in combination with movement for longer than 10 minutes.

pnharris02, 2025-01-22:
Some of this has been recently made possible through automations.  There is a step counter, which I use regularly, and a heart rate trigger.  Nothing specifically about intensity, and I haven't tried using the heart rate as an automation trigger (as I imagine it would vary too much), but if you knew your body's response to exercise then you could probably come up with a condition that would give you what you want with intensity too?