nightscout/AndroidAPS #1273: v3 Dash pod disconnects and will not reconnect

cheshire42, 2022-02-04:
Issue occurred at 15:11 PST (last communication with pod, reservoir @ 17.10U left
Build version d28e4b62e-2022.01.31-15:05

Samsung S21 Ultra (SM-G998W)
Android 12

In both cases, bluetooth headset was active. 
For 2022-02-01 Viber was being used for a voice call
For 2022-02-03 native phone app was being used for a voice call

For the 02-03, the low reservoir alarm was silenced

2 pods have lost communication with the phone, both with less than 20U left in the reservoir.
Have attempted: force quitting AAPS
Turn bluetooth off and on
Restart phone
Clear bluetooth cash (settings -> apps -> show system apps -> bluetooth -> clear cache, clear data)
2022-02-03 lost comms at 15:11 PST
2022-02-01 lost comms between 1100 and 1200 pst, log file may have been lost

[AndroidAPS._2022-02-03_16-28-37_.txt](https://github.com/nightscout/AndroidAPS/files/7999239/AndroidAPS._2022-02-03_16-28-37_.txt)

[AndroidAPS._2022-02-01_14-21-40_.txt](https://github.com/nightscout/AndroidAPS/files/7999238/AndroidAPS._2022-02-01_14-21-40_.txt)

ninelore, 2022-02-04:
Might be related to #1231

vanelsberg, 2022-02-05:
> 2 pods have lost communication with the phone, both with less than 20U left in the reservoir.

Apparently the pod was about to expire?
Wondering if maybe the pod just died (during the phone call when no comms was possible) due to battery depletion?

closed 2023-08-17 by MilosKozak

ninelore, 2023-08-17:
FYI I think we traced this on Discord to older pod lot numbers having bad BT Hard- or Firmware.

Those should be gone by now