nightscout/AndroidAPS #1274: 3.0.0.1 Problems: Small Screen, Prediction of active Insulin, reuse Profile Switch

McHoffi, 2022-02-04:
[AndroidAPS._2022-02-04_00-00-06_.2.zip](https://github.com/nightscout/AndroidAPS/files/8001718/AndroidAPS._2022-02-04_00-00-06_.2.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8001720/AndroidAPS.log)
[AndroidAPS._2022-02-04_00-00-06_.1.zip](https://github.com/nightscout/AndroidAPS/files/8001721/AndroidAPS._2022-02-04_00-00-06_.1.zip)
[image]
[image]
[image]

I use a Atom XL, it has a pretty small screen. In Treatments tab I can not delete some Treatments, I would have to rotate the screen, so I can delete treatments.

Also the Prediction of active Insulin is now missing, before the whole curve could be seen.

When doing a Profile Switch and you want to reuse it (does not matter if the same Profile is inserted, or you change the values somehow) it will give an error and set -6 hours.

When longpressing the old data takes long to load, I didnt test this now with these screenshots, so it may be not in the logs.

Time in logs is 09:39-09:40 (see screenshots).

Andries-Smit, 2022-02-04:
A quick response;
- treatment size has to be fixed, good you have a workaround and remove might not be needed too often. For smaller screens, a popup with the details might be a better option.
- Prediction issue might not be related to your screen size
- Profile issue seems a bug, we have to investigate
- What do you mean by "long-pressing" what data, what function? can you elaborate?

McHoffi, 2022-02-04:
By long-pressing i meant the expanding of the overview (BG-values, Treatments, Basal, etc.) on the Overview page on more hours. So going from 3 hours into the past and so on. I attach screenshots and the logs from now.

[AndroidAPS._2022-02-04_00-00-06_.5.zip](https://github.com/nightscout/AndroidAPS/files/8002502/AndroidAPS._2022-02-04_00-00-06_.5.zip)
[image]
[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8002503/AndroidAPS.log)

Philoul, 2022-02-05:
I made a PR #911 to (try to) fix graph rescale on long press (it shows quicker all the curves when rescale is done).
Unfortunatly it seems they are still some issues with it in history browser (I could not reproduce myself)...

Andries-Smit, 2022-02-24:
@McHoffi the treatment "Remove" buttons have been moved to the options menu in PR #1362 this makes the result in better screen usage.

closed 2022-02-24 by MilosKozak