nightscout/AndroidAPS #1275: DASH age/insulin level esthetic bug

robertrub, 2022-02-04:
@Andries-Smit You have simplified the information shown for EROS/DASH pods. Thanks :)

There is a small error on the interface (DASH in my case).

[image]

You have _Age:Insulin level_ when _Age:_ should be removed.

It would be nice to eliminate all information concerning the battery for DASH. The battery changes with each pod. 

[image]
[image]

The battery information for EROS should stay as it reports OL battery info.

Andries-Smit, 2022-02-04:
@robertrub it is fixed in PR https://github.com/nightscout/AndroidAPS/pull/1264
The battery is on my list, but first, this and a refactor of the status light has to be merged https://github.com/nightscout/AndroidAPS/pull/1200.
After that, it is a small change to do you request on battery

robertrub, 2022-02-04:
Thanks a lot :) Do we close this issue then ?

Andries-Smit, 2022-02-04:
It would be great if you can test the fix and comment in the PR, if it solves your issues then you can close this issue.

closed 2022-02-14 by MilosKozak