nightscout/AndroidAPS #1279: Nightscout crashes after updating to 3.0.0.1

illepidus, 2022-02-05:
AAPS Build: 479bfc410-2022.40-23.35
Nightscout version: 14.2.5 (current)

Nighscout hosted on herokku crashes after AAPS tryies to upload treatments to it.
It seems like AAPS is sending some null data:
```
2022-02-05T19:31:15.632726+00:00 app[web.1]: Got results null
2022-02-05T19:31:15.634857+00:00 app[web.1]: /app/node_modules/mongodb/lib/utils.js:106
2022-02-05T19:31:15.634858+00:00 app[web.1]:       throw err;
2022-02-05T19:31:15.634859+00:00 app[web.1]:       ^
2022-02-05T19:31:15.634859+00:00 app[web.1]: 
2022-02-05T19:31:15.634859+00:00 app[web.1]: TypeError: Cannot read property '_id' of null
```
Then I disable NS upload in AAPS, crashes never happen. Also AAPS is recieving data from Nightscout without any issues.

[androidaps-log.txt](https://github.com/nightscout/AndroidAPS/files/8008776/androidaps-log.txt)
[heroku-log.txt](https://github.com/nightscout/AndroidAPS/files/8008777/heroku-log.txt)

illepidus, 2022-02-06:
After hours of exploring my problem I found out that the Nightscout crashes only if both
- Upload Data to NS
- Receive therapy events

options are ON. So it's defenitly therapy sync problem.
Other NSClient synchronization options doesn't cause NS crash.

If someone face the same problem as me, try:
1.  AAPS->NSClient tab->check Paused.
2. AAPS->NSClient tab->NSClient Preferences->Synchronization->Turn off "Recieve therapy events".
_Other options might be in ON state depending on your needs._
3. AAPS->Maintenance->Reset Databases
_That would compleatly delete all treatments in AAPS database. **Be careful, CoB and IoB will become zero after it**. That could lead to insulin overdoze!_
4. AAPS->Exit
5. https://dashboard.heroku.com/apps/%YOUR_APP_NAME%/ -> More -> Restart all dynos
6. Launch AAPS
7. AAPS->NSClient tab->uncheck Paused.

illepidus, 2022-02-06:
NSClient 3.0.0 [] causes the same NS crush, but scince NSClient doesn't have NSClient tab->NSClient Preferences->Synchronization settings, it's impossible to turn off "Recieve therapy events".
So the only way to use NSClient for me and people having same issue is sticking with NSClient 2.8.2 [194549adc].

bbodong0814, 2022-02-07:
Is it ok if I turn off "Recieve therapy events", with "Upload Data to NS" turned on?

vagtil2005, 2022-02-07:
Have exactly the same problem, on 3.0 RC3, 3.0 and 3.0.0.1. Whenever there is a new calibration through xdrip, Nightscout crashes. Sometimes it takes 2 and 3 times to restart dynos in order to come back up. I am attaching the log files from heroku.

[heroku-logs-1644241845841.txt](https://github.com/nightscout/AndroidAPS/files/8015890/heroku-logs-1644241845841.txt)

MilosKozak, 2022-02-07:
try NS with this PR https://github.com/nightscout/cgm-remote-monitor/pull/7285

vagtil2005, 2022-02-11:
> try NS with this PR [nightscout/cgm-remote-monitor#7285](https://github.com/nightscout/cgm-remote-monitor/pull/7285)

I updated to patch-2 but the problem persists.

timocm, 2022-02-11:
AndroidAPS 3.0.0.1
Build: 3bbb57f33-2022.02.04-00:20
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.5
  xDrip+ G5 lastest night.   With calibration in xDrip+ nightscout crashes.      https://drive.google.com/file/d/11hoTq3sAGmJOxppT7MmnUkxiHAgHteyH/view?usp=drivesdk

timocm, 2022-02-20:
Did a deploy of NS with MilosZoxak-patch-2  (14.2.6) and an update to 3.0.0.1
Highlighted slow down uploads in settings  of he NSClient tab in AAPS.  Restarted dynos in heroku when the web socket crashed. It happened about 2 -3 times before there was no more crashing.
 Had to repeat the slow down upload setting and the restarting of dynos (2-4 times) each time I installed NSClient on a following phone (2 phones).
After a calibration form xDrip+ (G5) : crash and restarted dynos 3 times
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8103611/AndroidAPS.log)

timocm, 2022-02-21:
NS crashed at 7.14am  22 Feb with calibration in xDrip

Nightscout 14.2.6

AAPS 3.0.0.1
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8112239/AndroidAPS.log)

andrzej-aps, 2022-03-24:
> NSClient 3.0.0 causes the same NS crush, but scince NSClient doesn't have NSClient tab->NSClient Preferences->Synchronization settings, it's impossible to turn off "Recieve therapy events". So the [...]

There is a way to change synchronisation option also for NSclients take a look here: https://github.com/nightscout/AndroidAPS/issues/1486

closed 2022-05-19 by MilosKozak