nightscout/AndroidAPS #1282: 3.0.0.1 - Hypo TT replaces Active TT

haspden, 2022-02-06:
When you have a current Tenp Target such as Activity running, a hypo Temp Target will replace the Temp Target currently active. In previous versions I believe the previous temp target would have been re-enabled after a hypo Temp target?

Use case, out exercising with a tenp target ser for 240 mins of 8.0mmol . Hypo after 2 hours sets TT to hypo for 30 mins (in my case). After 30mins TT does not return to 8.0 activity target but instead the default profile target.

[image]

MilosKozak, 2022-02-06:
this behavior is expected

closed 2022-02-07 by MilosKozak

haspden, 2022-02-07:
@MilosKozak In my eyes not desired or safe... (with safety first and all being the name of the game)

If a temporary target overnight has been set for a specific reason (such as 8.0mmol post exercise in my case) when entering a carbs a Hypo TT is automatically added, it should not remove my 600 minute long temp target, but instead take precedence over this for 30 minutes as the most recent Temp Target before returning to the original / originals

Andries-Smit, 2022-02-07:
When the user sets a new temp target, it is expected it replaces the old one. But in this case, the systems change the temp target without the user noticing, which is undesired in this case.  The hypo TT is set by the automation. Can we find a way to automate the restore of the old TT in another way?

haspden, 2022-02-07:
It's not even necessarily by automation @Andries-Smit as hypo Temp targets are set by default when low and adding carbs thus overriding targets set for a specific reason...

An alternative solution to my higherarchy method proposed above would be perhaps to prompt if a temp target exists if the user wanted to add it in addition (perhaps then add a temp target scheduled for future when the new one expires) or replace it...

Koeppchen, 2022-11-01:
From a user interface view it would be good to have the stacking behaviour just like the profiles. And i also lost my sports target due to setting a shorter hypo target. What happend to @haspden also happend to me during sports and having smb's delivered during sports due to lost temp targets is not good.
This is the reason why i would love to have stacking for  temp targets.