nightscout/AndroidAPS #1283: DASH : Min required units wrong

robertrub, 2022-02-06:
@MilosKozak or @avereha This bug was corrected, I didn't understand why it came back in master 3.0.0...

[image]

The minimal units needed to fill the EROS and DASH pods in 80U and not 85U. I have tested this several times filling the pod with a insulin pen.

Andries-Smit, 2022-02-06:
85 was the experience shared discord to a conversation: https://discord.com/channels/629952586895851530/680407341820280863/790137451611553823

robertrub, 2022-02-07:
Nope, @teleriddler is wrong. I have at least 3 pen fill up of the pod and the pen can only inject 80U at one time and each time, after the 80U was injected, the pod beeped. 80U was also the number given when I was learning to use the pod.

Andries-Smit, 2022-02-07:
You can be right when it beeps,  however, the official Dash documentation is leading 
https://www.omnipod.com/sites/default/files/2021-08/AU%20Omnipod%20DASH%20SYSTEM%20%20User%20Guide%20PT00032.pdf page 48

> Note: The Pod must contain a minimum
> of 85 units of insulin in order to function.
> The Pod beeps twice after it has been filled
> with 85 units of insulin. If you have filled
> the Pod with more than 85 units and still
> do not hear the two beeps, please call
> Customer Care.

A reason, not the required minimum in text not to be equal to the beeps.; users might be mistaken by a rounding/reading error when measuring the insulin. better by 5 up, than 5 to low.

closed 2022-02-07 by MilosKozak

teleriddler, 2022-02-08:
> Nope, @teleriddler is wrong. I have at least 3 pen fill up of the pod and the pen can only inject 80U at one time and each time, after the 80U was injected, the pod beeped. 80U was also the number given when I was learning to use the pod. 

I was basing this off the official Insulet documentation for EROS pods but it should be the same for DASH pods.  

I agree that 80 units triggers the beeps but you can put in more insulin afterwards.  Some people take more insulin for the 3 day span so while 80 units is the minimum amount needed to activate the pod more can be injected for those that require it.

TR