nightscout/AndroidAPS #1284: The current profile is not visible in the profile editor

JimmyFoks, 2022-02-06:
In the profile editor (and even after touching the name of the active profile on the main screen) I can't select the profile that is currently active. I don't see any of the two newly created local profiles today (manually via mobile phone).

AndroidAPS 3.0.0.1; Android 1
[image]
[image]

MilosKozak, 2022-02-06:
if you set profile and then delete it in profile it will be still running

MilosKozak, 2022-02-06:
you can still clone running profile from treatments

JimmyFoks, 2022-02-07:
The profile was never deleted by me.
And as another interesting fact - when I enter the profile editor, I change something in the displayed profile (not the current one on the main screen) and then I touch RESET, the profiles are displayed - see the video.

https://user-images.githubusercontent.com/34617797/152727427-0821f5a8-8191-414e-93b7-5bfad3b2688d.mp4

MilosKozak, 2022-02-07:
this seems to be caused by synchronization with NS (different profiles received from NS during editing)
review all profiles to be without error, do small change and save ....
then you should see the same profiles in NS

JimmyFoks, 2022-02-07:
Okay.

closed 2022-02-09 by MilosKozak