nightscout/AndroidAPS #1288: Omnipod and DASH tabs force closes AAPS

zeppo33, 2022-02-07:
When alarm for low reservoir or end timer occurs, pushing the tab for omnipod or dash (I tried both types of pods with same error) simply force closes the app. It is impossible to "ack alarms" or change the pod. 
So far my solutions is resetting AAPS, importing old import and starting over with a new pod while physically deactivating the old one. I have been searching around for solutions and have tried reinstalling, resetting and even rebuilding the app package.
The last occurrences happened several times between 01.00 - 02.24, but i was resetting and reinstalling several times in the timespan - wondering if log files are useless then?

Build v.: 3.0.0.1 (and 3.0.0)

The issue does not occur on v.2.8.2

Phone: Samsung SM-G981B/DS, Android 12
[AndroidAPS._2022-02-06_02-27-07_.0.zip](https://github.com/nightscout/AndroidAPS/files/8014520/AndroidAPS._2022-02-06_02-27-07_.0.zip)

MilosKozak, 2022-02-07:
Nothing in log. Is it replicable?

zeppo33, 2022-02-07:
> Nothing in log. Is it replicable?

I guess it is, but I will have to wait until the Pod triggers an alarm for either. I am trying to adjust the alarm time for pod depletion to suit my needs for a new log file. I guess I can try letting it force close a couple of times and then quickly download the latest log file to make sure its the correct one. I will hopefully return with it in ~24h

m-scha1337, 2022-02-11:
@zeppo33 might be a long-shot but could you try and change your phones system language from danish to english?
Somebody on facebook has raised a similar issue and their system language is also set to danish.

Also: Do you know how to record a logcat?

zeppo33, 2022-02-12:
@m-scha1337 I looked for the other guy, and it seems you were right. So I did some more digging to try and narrow it down:
1. The omnipod/dash tabs seem to force close AAPS when the pod has <50U left (could be something with this translation string?)
2. Changing language to English in AAPS does not change names of the tabs, thus not solving the problem (can not tell if it changes the language IN the tab)
3. Changing phone language to English changes all translations back to original and the problem disappears.

zeppo33, 2022-02-12:
> * The omnipod/dash tabs seem to force close AAPS when the pod has <50U left (could be something with this translation string?)
> 
> * Changing language to English in AAPS does not change names of the tabs, thus not solving the problem (can not tell if it changes the language IN the tab)
> 
> * Changing phone language to English changes all translations back to original and the problem disappears.

Issue with this description is now confirmed by another user in this facebook thread: https://www.facebook.com/726516084/videos/3057617701166751/

m-scha1337, 2022-02-12:
@MilosKozak I think this warrants the "bug" label

MilosKozak, 2022-02-14:
i already fixed some crashes related to wrong translations. Could you test 3.0.0.1-dev?

m-scha1337, 2022-02-15:
> i already fixed some crashes related to wrong translations. Could you test 3.0.0.1-dev?

well @MilosKozak, the thing is.. I was going to test this in 3.0.0.1-Release, forcing AAPS into using using danish via root shenanigans (zygisk, lsposed and "app settings reborn", changing 'locale' to danish). My end goal was to check logcat for anything suspicious..
I had already set up pretty much everything, but in the end i had to give up because i can't really think of a way of faking a real RileyLink connection and faking a real Omnipod dash connection that reports <=50 IU remaining.. 
Since i'm not willing to sacrifice a real pod for this test (this is a burner phone so i would need to sacrifice a pod upon switching back to my main phone), i abandoned this idea.

Therefore I am asking you if there is a way of faking a RileyLink connection and faking an active Omnipod dash, configured to report <=50 IU remaining, without using the "Virtual Pump" driver interface..
Even if this issue has already been fixed, i would like to know for future testing :)

zeppo33, 2022-02-18:
> i already fixed some crashes related to wrong translations. Could you test 3.0.0.1-dev?

@MilosKozak I have never tried anything but the "master" version. Can i just build, upgrade and use straight out of the box? Can i just as easily revert to 3.0.0.1?

m-scha1337, 2022-02-19:
> > i already fixed some crashes related to wrong translations. Could you test 3.0.0.1-dev?
> 
> MilosKozak I have never tried anything but the "master" version. Can i just build, upgrade and use straight out of the box? Can i just as easily revert to 3.0.0.1?

Take this with a grain of salt because frankly, i'm not 100% sure since i don't really do kotlin/java but: I think you will have a tough time going back from 3.0.0.1 dev to 3.0.0.1 master as android generally does not allow app downgrades..

I believe in order to go back from the dev build to the master branch build, you'd need to completely uninstall the AAPS dev .apk, then re-install the AAPS master .apk.

I would _also_ be careful not to import settings generated in a more recent version of AAPS, into an older version of it. As i scrolled through `VersionCheckerPlugin.kt`, i found that `PrefFileListProvider.kt` (which i assume is responsible for import/export of settings) calls `versionCheckerUtils.versionDigits()` and compares app version numbers with the app version attached to the settings file and the current version (which in this case could both just be '3.0.0.1' but idk).
As I understand it, importing different **minor** version settings _should_ be fine, but there's no guarantee of course..

---

Then again; i could be totally wrong about all of this.. Those are just my two cents on the matter.
glhf.

Andries-Smit, 2022-02-22:
Wrong translations should no longer cause crashes and should fall back to the English text and report an issue via Crashlytics. #1329

zeppo33, 2022-03-05:
> Wrong translations should no longer cause crashes and should fall back to the English text and report an issue via Crashlytics. #1329

So this should be working again from the the next master version available after 3.0.0.1?

closed 2022-04-07 by MilosKozak

danielhers, 2022-05-29:
This is happening to me on 3.0.0.2. Same problem and same conditions.