nightscout/AndroidAPS #1291: BYDO-Aaps communications

crizz6396, 2022-02-08:
Reporting bugs
Randomly BYODA & AAPS stop communicating, I can not identify what is interfering with the communication.  Attached is the log files.

[Log Export 2-8.zip](https://github.com/nightscout/AndroidAPS/files/8026535/Log.Export.2-8.zip)

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

crizz6396, 2022-02-09:
This is only happening when I am at work.  This is nothing any sense.  Can putting my phone on Vibrate, wifi network cause issue?

MilosKozak, 2022-05-27:
reason not known

closed 2022-05-27 by MilosKozak