nightscout/AndroidAPS #1293: Permanent warning next to BG number

dkmsoft, 2022-02-08:
First off, thanks so much for all the work put in to AAPS.
I have an issue with 3.0.0.1 and the new warning icon next to the BG number. My understanding from what I read from Milos is that it shows when recalculated data is used, ie when data is recalculated to be in 5 minute intervals. My data is coming from xDrip and is sent every 5 minutes (see screenshot). So why is the warning icon permanently showing?
It is very disquieting to see this permanent warning when everything seems fine and is working - not good UI practice in my humble opinion.
[image]

MilosKozak, 2022-02-09:
AAPS uses around last 30 hours of BGs for calculations. triangle means the irregularity outside tolerance appeared within this period.
OAPS is designed on top of 5 min BG data and with recalculation there could be delay in response. therefore the warning

closed 2022-02-09 by MilosKozak

dkmsoft, 2022-02-09:
Thanks for response. Please consider giving option to dismiss the warning - I think a couple of missed readings in last 30 hours is not material and warrants this. A number of others have complained about this that I have seen. Clicking on icon once acknowledged should dismiss it or at least add an option in preferences not to show it.

MilosKozak, 2022-02-09:
missing BG itself doesn't trigger exclamation mark. it must be something like BG 4 min from previous etc

maheini, 2022-02-09:
Got that as well - since 2 months now. It's perfectly working, looks a little strange and therefore a useless warning for me.
Setup: Libre 2 patched -> Xdrip -> AAPS v. 3.x

a thought about the cause:
Is it because xdrip does not calculate the values from the Libre 2 exactly every 5 min (which is the case)?

dkmsoft, 2022-02-10:
Milos, thanks for explanation. I understand this is not a bug, just strict requirement placed on interval of BG values.
But please try appreciate my point. I think this happens for all Libre users (and there are many) and what it means is that all
these users have a 24x7 warning displayed on the main screen and THERE IS NOTHING THEY CAN DO ABOUT IT. Not all can switch to Dexcom. (I tried Dexcom for over a year and does not work well for me and more inaccurate than Libre in my experience). So this means all these users have to learn to ignore a warning that they see all the time. Everything works fine, just has it has done for all previous versions.
All I am asking is to be able to dismiss this warning once acknowledged. That would be correct user interface when the user  can't do anything about the issue and actually does not need to. (I have been developing software for over 48 years with tens of thousands of users worldwide). All your Libre users will thank you for that.

Please also understand I really appreciate what you have done - an incredible piece of software for which I am very grateful.

maheini, 2022-02-10:
@dkmsoft It's basically not a real problem and Milos and all the other developers have lot's of other important things to do as well. I really appreciate all the effort put into AndroidAPS and the fact it's made available for everyone.

I made a quick fork (as a temporary workaround) with disabled Warning. Please keep in mind that you are using it at your own risk. It's based on AndroidAPS 3.0.1, I only disabled the yellow warning. Here's the fork: [Link to the AndroidAPS Fork without warning](https://github.com/maheini/AndroidAPS/tree/AndroidAPS_3.0.0.1_disabled_ic-baseline-warning). Feel free to build your AndroidAPS from this fork.

Maybe I'll even find the time & solution to create a real fix.

dkmsoft, 2022-02-10:
@maheini Thank you very much.