nightscout/AndroidAPS #1305: 2.8.2.1 crashes when switching the phone from flight mode to normal

TinoLL, 2022-02-11:
Using my Blackberyy Key2 LE (Android 8) over night in flight mode. Bluetooth and NFC are on. 

When switching back to normal mode AAPS disappears. No freezing, nothing but just swiches off. 

AccuCheck Combo, Libre 2 with patched app, XDrip, Ruffy, Buikd fb9325384-2022.02.02-17:25

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the 
[AndroidAPS_LOG_1644582903973.log.zip](https://github.com/nightscout/AndroidAPS/files/8048677/AndroidAPS_LOG_1644582903973.log.zip)
Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the 
[AndroidAPS_LOG_1644582903973.log.zip](https://github.com/nightscout/AndroidAPS/files/8048662/AndroidAPS_LOG_1644582903973.log.zip)
app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-02-13:
there will be no updates to 2.8.X version anymore. Test it in version 3 and open new issue if you can reproduce it.
My test phone didn't crash on switching to flight mode. But i didn't  have Combo connected

TinoLL, 2022-02-14:
3.x doesn't run under Android 8. So what to do now? Reporting that 3.x doesm't run under Android 8 as bug?

MilosKozak, 2022-02-14:
even Android 8 is very old (we have 12 now). We kept 2.8 working to give users time to upgrade phone (or flash Lineage OS etc) and make 3.0 running. But we are not company that can dedicate someone to maintain old versions. Sorry

MilosKozak, 2022-02-14:
AAPS 3.0 has more than 50% code new. It would mean maintain 2 different versions, basicaly do everything twice

TinoLL, 2022-02-14:
I fully understand. And I appreciate your great work. There are workarounds for the crash: restarting Ruffy and then AAPS (if someone else asks....). And in the future I will have a newer Android version. For the moment I can live with it as it ist. Thank you.