nightscout/AndroidAPS #1308: NSClient 3.0.0.1.           The time display is massive on the top right side: can this be made smaller or be removed so that the deltas can be moved over to the right and the BG value can be moved to the right allowing the BG value to be completely visible when there is a double digit BG value or if the font size of the phone is increased?

timocm, 2022-02-11:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

mick711, 2022-02-12:
Could you please edit and add title 😉

timocm, 2022-02-13:
NSClient 3.0.0.1. The time display is massive on the top right side: can this be made smaller or be removed so that the deltas can be moved over to the right and the BG value can be moved to the right allowing the BG value to be completely visible when there is a double digit BG value or if the font size of the phone is increased? #1308

MilosKozak, 2022-02-13:
can you provide screenshot?

timocm, 2022-02-13:
[image]
With present font settings : no space for a BG > than 9.9

timocm, 2022-02-13:
[image]
With a setting of a larger font size

timocm, 2022-02-13:
[image]
With a setting of a larger display size

MilosKozak, 2022-02-13:
what is this phone?

timocm, 2022-02-14:
Galaxy note20 ultra

kristinsmotz, 2022-02-14:
I get the same if I increase font size in phone settings larger than the third available size (dot on the scale line). Samsung Flip 3. I just use the 2nd or 3rd sizes. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MilosKozak, 2022-02-14:
> what is this phone?

and you font size?

timocm, 2022-02-14:
I was  hoping for less space for the time and a more centralized position of the BG value
[image]

Andries-Smit, 2022-03-12:
@timocm I scaled down the text size in PR #1436 could you validate this works for you well for your device?

PR #1436 (2022-03-14): Fix nightscout client big time text
When running the nightscout client, the time is shown, this was too big and made part of the UI unusable
With the PR the font size is made smaller.

[image]

Fixes #1308
closed 2022-03-16 by MilosKozak