nightscout/AndroidAPS #1309: [AAPS 3.0.0.1] Frequent pump unreachable errors with Orangelink and Emalink

happygirlnyc70, 2022-02-12:
Pixel 4a 5g Android 12
AAPS build479bfc410

On three occasions now either the orangelink or emalink has disconnected from AAPS

Log is from 11Feb noticed issue around about 9 AM

I received an alert that my pump was unreachable.
Steps I took to fix the issue (play protect is disabled - verified that it is off):
1) Refreshed pod status
2) Scan for link
3) Remove link and re-scan
4) switch to virtual pump then back to eros to restart pump driver
5) restart AAPS
6) restart phone

Both links are able to connect to to nRF connect. Sometimes the link will connect but it fails to refresh. Other times it stays in the initialization step

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
[AndroidAPS_LOG_1644588029325.log.zip](https://github.com/nightscout/AndroidAPS/files/8052413/AndroidAPS_LOG_1644588029325.log.zip)

harentius, 2022-02-12:
I observe something similar

robertrub, 2022-02-12:
I had made a wiki on how to restore a dead connection in Discord AAPS wiki. It is dated may 19th(?), 2020.

Normally, the connection to OL code was perfected and this problem was solved...

Next time it happens, try the solution in the wiki.

Which OL firmware are you using ?

happygirlnyc70, 2022-02-13:
@robertrub I did shut off the orange link but I didn't for e stop aaps. I will try that next time. I'm using fw 3.2. I also turned off adaptive battery. I do have optimization turned off for aaps but I'm wondering if adaptive battery is impacting Bluetooth.

robertrub, 2022-02-13:
Doing it on order is a must ;)

Yes, you need to stop battery optimization for BT also.

Try the OL firmware version before v3.2 (from memory v2.5). It was working better. 

Once again, this was supposed to have been solved. So keep this issue open and add info here.

happygirlnyc70, 2022-02-14:
I have not had any problem since I reported the issue.  @harentius if you are still having similar problems please add to this

happygirlnyc70, 2022-02-17:
The orangelink disconnected again today.  The instructions from the discord wiki fixed it. Here's the log 
[AndroidAPS_LOG_1645032158112.log.zip](https://github.com/nightscout/AndroidAPS/files/8085283/AndroidAPS_LOG_1645032158112.log.zip)
t

Technixx, 2022-04-08:
Hi, i have a similar problem.

My son is using AAPS v3, on Unihertz Jelly2 (Android 10 i believe) , I have tested on Orangelink Pro and original with fw 3.2. 
The pod looses connection a lot (pod unreachable) and it also fails multiple times during activation with fault:
049 | 0x31 | Incorrect pod state for command

I have seen an incline in errors when changing omnipod batches. 

With this batch I have detected more problems:
https://photos.app.goo.gl/3haYCFYqrHjLwwoG6 

With this batch, not so much
https://photos.app.goo.gl/XryfViPLN6BKPKNH9

happygirlnyc70, 2022-04-13:
@MilosKozak I believe we can close this ticket as well. @Technixx since you have opened your own ticket I am closing this one.

Technixx, 2022-04-13:
> 

Quite ok with me. 👍