nightscout/AndroidAPS #1313: Error when updating the basal profile

Dragancokan, 2022-02-12:
Hi,
since update to AAPS 3.0.0.0 and 3.0.0.1 I have again and again "Error when updating the basal profile" but only with the Automatisirung and that not every time. 
[image]

When I manually change profile this does not happen. I use Android 12 smartphone and Dana RS FW 3.
Does anyone else have problems like this? Any idea what I can do about it? Except turning off automatism.

Greetings Dragan

MilosKozak, 2022-02-13:
did you verify max basal in pump acording your profile switch % ?

Dragancokan, 2022-02-13:
The problem doesn't always occur. If I confirm with OK then everything is ok the next time I change.
I have a Dana RS. I had max basal U/H set to 10 and now set it to 16. That's max.
Since I have U200 insulin, I think that 10, ie 20IE, was enough. Will test it now with new setting max basal U/H 16

Dragancokan, 2022-02-14:
It's not because of that. Just happened again. Here is the current log
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8061238/AndroidAPS.log)
.

MilosKozak, 2022-02-14:
```
14:38:22.374 [Thread-34411] D/PUMPBTCOMM: [BLEComm.sendMessage():772]: >>>>> BASAL__SET_PROFILE_BASAL_RATE B2 66  00 4B 00 4F  00 5A 00 67  00 79 00 8C  00 8F 00 78  00 63 00 53  00 4E 00 4D  00 4D 00 4D  00 4F 00 58  00 64 00 6C  00 67 00 5E  00 56 00 51  00 4F 00 4D  00 
14:38:27.427 [Thread-34411] W/PUMPBTCOMM: [BLEComm.sendMessage():832]: Reply not received BASAL__SET_PROFILE_BASAL_RATE
14:38:27.428 [Thread-34411] D/PUMPCOMM: [DanaRSPacketBasalSetProfileNumber.<init>():14]: Setting profile number 0
14:38:27.429 [Thread-34411] D/PUMPBTCOMM: [BLEComm.sendMessage():772]: >>>>> BASAL__SET_PROFILE_NUMBER B2 64  00 
14:38:32.480 [Thread-34411] W/PUMPBTCOMM: [BLEComm.sendMessage():832]: Reply not received BASAL__SET_PROFILE_NUMBER
```
It's some kind of pump issue. Pump did't respond to the command but 30 sec later the same command was succesfull. AAPS responded correctly: reported failed command.

Sorry there is nothing I can do

closed 2022-02-14 by MilosKozak

Dragancokan, 2022-02-14:
Should I contact Pump Support? Have the pump replaced?