nightscout/AndroidAPS #1316: Feature request: Show BG and arrow on phone unlock screen

robertrub, 2022-02-13:
xDrip has a nice feature, show BG and arrow on the unlock screen of phone. You just wake up the screen to see your BG.

It would be nice for BYODA users if AAPS had something similar.

[image]

Alamo04, 2022-02-13:
Would be a nice to have feature, for all users without XDrip. 

For now I use Boyda and XDrip only for my Watch and this Lock Screen feature. :)

MilosKozak, 2022-02-13:
i don't know even how to enable it

robertrub, 2022-02-14:
[image]

Andries-Smit, 2022-03-08:
The question is, should this be implemented in BYODA, and not AAPS?

Logic and most of the code can be reused from https://github.com/NightscoutFoundation/xDrip/search?p=1&q=number+wall

However, BYODA has an option to support both xDrip and AAPS config
> Would you like to broadcast your glucose readings to other apps? G6 ONLY
> [ ] Yes, broadcast to both xDrip and Android APS.

There is also an option for Always-on Display for Wear OS watches too.

closed 2022-07-25 by robertrub