nightscout/AndroidAPS #1322: Dana-R incorrect message when occlusion

dkmsoft, 2022-02-15:
When the Dana-R has an occlusion, the following error message is displayed in AAPS:
"Data is coming from different pump. Change pump driver to reset pump state."
This has been verified over a number of occlusions. I think also happens with other pump initiated actions like priming.
Issue did not happen in versions prior to Rel 3. Recent log file attached.
[AndroidAPS_LOG_1644885484809.log.zip](https://github.com/nightscout/AndroidAPS/files/8064885/AndroidAPS_LOG_1644885484809.log.zip)

MilosKozak, 2022-02-20:
didn't you flash pump recently to v2? SN doesn't match BT name
i'd recommend set proper name again, forget and repair the pump in android and thes set again in AAPS

dkmsoft, 2022-02-20:
The pump has been on v2 for about 5 years now. Also, the SN in BT list does match the one set in AAPS (else I
doubt all other pump connections would work). It worked fine for Rel 2.2 - Rel 2.8.2, issue only occurs when
using Rel 3.0.0.1. Fortunately all other driver functions and handshaking work perfectly.
If you did work on Dana-R driver from Rel 2.8.2 to Rel 3, I suspect something was changed to cause this.

MilosKozak, 2026-10-15:
Fixed in 7d3e2b0ab5 (dev).

On an occlusion (and also on low battery, pump shutdown and pump errors) the pump sends an error message, and AAPS writes a note about it to the treatments. That note used the pump type that comes from the pump firmware, while all other records use the type of the selected driver. On a Dana-R these two can differ: the firmware reports Dana-R or Dana-R v2 depending on its protocol version. So the note had a different pump type than the registered pump, AAPS rejected it as data from another pump and showed "Data is coming from different pump". The occlusion note was also missing in the treatments.

Now the note uses the type of the selected driver. The same commit also fixed the "approaching daily limit" note on Dana-R and Dana-R v2, which was written with the Dana-R Korean type and was rejected the same way.

closed 2026-10-15 by MilosKozak