nightscout/AndroidAPS #1323: Obj. 3 - Profile switch Question to be corrected

xJoe, 2022-02-15:
The following issue was started by https://github.com/klalle in AAPS docs and should be delt by app team as the question text must be updated.
 https://github.com/openaps/AndroidAPSdocs/issues/1126

This is something that tricked me when going through the objective 3.
If you set a 90 % profile, the example states that the ISF is devided by 0.9, which is not a 10 % increase last time I checked...
And the example image in the docs is also wrong (2.2/0.9 != 2.4) its more like 2.44444. (yes you have 2 digits on some numbers). but an 10 % increase would be: 2.2+0.22=2.42.

[image]

Yes it's close to a 10 % increase, but lots of us are engineers, and we will have a hard time with this question since you'd have to answer wrong to get it right! Probably more than me that had to check the source-code to get this right ;)
image

[image]

To prove the point, lets take something larger: if you set a profile switch of 60 % (not even sure its posible) and an ISF of 10.
10/0.6 = 16,6667, but a 40 % increase of 10 is 14.0!

Or am I misinterpreting this on some way?

[image]

ga-zelle, 2022-03-03:
Here we go with an example: 
Let's say the ISF is 40 and we change profile to 80%. We would expect a change in sensitivity of 20%.
The adjusted ISF now is 40/0.8 = 50
Say we want to have a bolus that brings us down by 40mg/dl. Originally we needed 40/40=1U f insulin.
With the reduced profile we need 40/50=40/(40/0.8))=40/40 * 0.8=0.8U of insulin. That is 20% less than at full profile!

So, it is not the ISF changing by 20% but the sensitivity changes by 20%. ISF is somehow just the inverse of what mostzthink it is. It is like an engineer from Europe discussing a 20% improvement in fuel consumption with his colleague from the US. The one thinks in l/100km, the other in miles/gallon. Because one metric is more or less the inverse of the other they never find a common ground.

ga-zelle, 2022-03-03:
> but the sensitivity changes by 20%

To be more precise:  the insulin needed changes by 20%.

MilosKozak, 2022-05-27:
we could change it to "about 10%" but i'm not sure if it's a worth force retranslating strings in all languages

closed 2026-10-15 by MilosKozak