nightscout/AndroidAPS #1326: AAPS v3 - after loosing an omnipod to an error (empty reservoir), get same error when trying to start new one

SusiDoesntLikeUsernames, 2022-02-16:
I'm using AAPs 3.0.0.1, the Omnipod Eros system and have the connection through an EmaLink nano. 

At 7.56pm (Feb 15) the pod that had been running for 3 days (pod A) ran out of insulin (0 Units left) while I was giving myself a bolus. It did do it's omnipod thing (beeping really loud and really annoying) which I've stopped by using the pencil in the little button/whole trick. 
After that I've deactivated the pod (pod A) and when this did not establish a connection dismissed it (not sure if this is the right term in english - "pod verwerfen" in German. Next, I've started a new pod (pod B) as usual only to get the an error "024 Reservoir
_empty". I've tried a bunch of times, restarted the app, disconnected and reconnected the Riley link, still got the same error. I've tried again the initiating process with the same pod again, and interestingly got the clicking noises from the pod indicating that id did in fact connect and prepared something. But still the same error message, so so I've also deactivated pod B, which (if I remember correctly) worked without any error and the pod did it's little beeping telling me something happened.

I restarted my phone and disconnected and connected the Riley link before trying again with a new pod (pod C) - same spiel as before, when connecting I got the "024 Reservoir_empty" error. Now I removed all old pods (A and B) out of reach (in case my phone still connected to them) and switched out my ema link nano to an ema link micro had as a reserve and restarted my phone again. Trying again with pod C I got the same error, so I tried to deactivate that pod, did not get a connection, so I dismissed it (that lower right button below disconnect pod, again, not sure about the english term).

Connection to the next pod (pod D) worked. Not super sure which of all the things I've tried helped.  
Attached two logs, first one ("AndroidAPS._2022-02-15_20-27-04_.0.txt") is  7:13pm to 8:28pm, second one (AndroidAPS.log) is 8:28 to 9:13pm. problem got resolved around 8:55 pm. 
Thank you!

[AndroidAPS._2022-02-15_20-27-04_.0.txt](https://github.com/nightscout/AndroidAPS/files/807604
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8076046/AndroidAPS.log)
5/AndroidAPS._2022-02-15_20-27-04_.0.txt)

SusiDoesntLikeUsernames, 2022-02-16:
okay, looks like the first log did not upload properly. renaming it in hopes that this helps
[AndroidAPS2.log](https://github.com/nightscout/AndroidAPS/files/8076084/AndroidAPS2.log)
:

SusiDoesntLikeUsernames, 2022-02-16:
Update:
Pod D failed with the same error as the other ones (024 Reservoir_empty) at 10:07pm. it did so when I was trying to pause the loop (in a failed attempted to get it to accept that a basal rate of 0 is not great idea after not getting any insulin for 3 h and during dinner cause of failed pods).
Interestingly the  app displayed very strange stats about the pod - it claimed the date was Feb 19 3:55 am when it was Feb 15 22:10. At this time (Feb 19) Pod D the will have pretty much exactly the same age as the initial Pod A on Feb 15 22:10.  So I guess this is related. I've attached a screen shot -  there are other strange things as well. 
[image]

On a good note, I just left the pod on my arm anyway cause basal rate is in there and I'm tired, and now it's working again... this was maybe around 10:35pm. Not super sure. I've attached the newest logs. "AndroidAPS._2022-02-15_20-27-04_.1.log" shows 20:28 to  22:15, androidAPS.log is 22:15 to 22:38

[AndroidAPS._2022-02-15_20-27-04_.1.log](https://github.com/nightscout/AndroidAPS/files/8076337/AndroidAPS._2022-02-15_20-27-04_.1.log)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8076339/AndroidAPS.log)

vanelsberg, 2022-02-16:
(Btw: Check you NSClient settings. It seams to download a lot of treatment data from NS.
Set it to "Upload data to NS", all other off.)

* After "Discarding" the Pod and before activating new, did it say "No Pod active"?
When activating current Pod, was this succesfull? Any errors?

* Going on the screen print it is looks like it is showing old Pod status.
Could you double-check if RL is connected to the Pod you think it is? (Check Pod numbers displayed with the active Pod?)
In case of doubt, make sure none of the problem pods is reachable and do a rescan/reconnect op RL config. Once connected, press the "Refresh" button.

* Check your build commit (is it latest on the 3.0.0.1 branch?).
(I remember early release having a bug that was fixed shortly after releasing 3.0.0.1?)

vanelsberg, 2022-02-16:
> Isn't that button also greyed out when another pod is still connected?

Yes, not after discarding the Pod (though the Pod itself can still be "active" - AAPS does not have it registerd anymore - that at least id the idea)

> it says the build is 479bfc4

Yes. That is latest 3.0.0.1. You are Ok.

Along the way of trying to solve the problem. did you import settings by any chance?

vanelsberg, 2022-02-16:
Looking at you logs did not help (me).
@avereha could you maybe have a look at this problem?

SusiDoesntLikeUsernames, 2022-02-16:
Thanks for the help!

> Along the way of trying to solve the problem. did you import settings by any chance?

No, I did not. However I did have that check mark to get profiles from NS and did restart the app. Not sure if that is relevant.

SusiDoesntLikeUsernames, 2022-02-16:
I can also reproduce the error by taking my old pod (pod A) out of the freezer and unlink/relink the RL. 
While in the freezer things worked perfectly fine.

Do you have any specific tests in mind that you would like me to do while the error is occurring? @vanelsberg @avereha

SusiDoesntLikeUsernames, 2022-02-16:
Okay, I've done some testing.
I took a screen recording so people can see what happens: https://photos.app.goo.gl/7RX76QUjX7MkvzJg7
It's super long and the audio is the worst, but it's what I got. The file is also to big to embed, hence the link. 
Log of that time period is also attached.

1) I started out with the old pod in a tin can in the other room. In that case updating it and setting a new temporary rate does work as intended.

at **minute 1:11**  (=11.11am)  I disconnect the RL and get up to **get the faulty pod** from the other room (that's the loud noises)

2) old pod next to my phone. reconnecting RL results in the warning. tried to get the pod to test beep, got an error, showing the RL status and history.

at **minute 2:24** (=11.13am) I **return the old pod to the other room** into a tin can.

3) updating gives the no-error screen, test beep, works. showing the RL status and history. Changing temp basal rate works 

at **minute 4:03** (=11.15am)  **bringing the old pod back** to the phone. disconnecting and reconnecting the RL. 

4) get the error again when updating the omnipod screen. test tone is working (new pod is beeping) as a result screen is back to normal (no error)
  
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8081117/AndroidAPS.log)

SusiDoesntLikeUsernames, 2022-02-16:
one more log - this time I did not trigger the error but it just happened before I had the chance to return the old pod to the freezer. Should be the last entry in this log (11:49:58.303 is where I've stopped the alarm, no clue what triggered it).

[AndroidAPS_error without forcing it.log](https://github.com/nightscout/AndroidAPS/files/8081401/AndroidAPS_error.without.forcing.it.log)