nightscout/AndroidAPS #1327: AAPS 3.0.0.0.1 OpenHumans Upload doesn't work anymore

lutzlukesch, 2022-02-16:
Setup: Galaxy Z Flip, AAPS 3.0.0.0.1, Omnipod Dash
I did remark this as my OpenHumans Data where no longer uploaded to OH  and did look into the Logs:
07:40:10.651 [DefaultDispatcher-worker-1] I/OHUPLOADER: [OpenHumansWorker.doWork():28]: Starting upload
07:40:10.806 [DefaultDispatcher-worker-1] E/OHUPLOADER: [OpenHumansWorker.doWork():34]: OH Uploader failed
java.lang.UnsupportedOperationException: Tried to obtain display from a Context not associated with  one. Only visual Contexts (such as Activity or one created with Context#createWindowContext) or ones created with Context#createDisplayContext are associated with displays. Other types of Contexts are typically related to background entities and may return an arbitrary display.
	at android.app.ContextImpl.getDisplay(ContextImpl.java:2602) ~[na:0.0]
	at android.content.ContextWrapper.getDisplay(ContextWrapper.java:1022) ~[na:0.0]
	at info.nightscout.androidaps.plugin.general.openhumans.OpenHumansUploader.uploadDataPaged(OpenHumansUploader.kt:237) ~[na:0.0]
	at info.nightscout.androidaps.plugin.general.openhumans.OpenHumansUploader.access$uploadDataPaged(OpenHumansUploader.kt:47) ~[na:0.0]
	at info.nightscout.androidaps.plugin.general.openhumans.OpenHumansUploader$uploadDataPaged$1.invokeSuspend(Unknown Source:19) ~[na:0.0]
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33) ~[na:0.0]
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106) ~[na:0.0]
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571) ~[na:0.0]
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738) ~[na:0.0]
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678) ~[na:0.0]
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665) ~[na:0.0]

I now use a Workaround: In the OpenHumansUploader.kt class on 236 I did comment out the if statement:
   val displayMetrics = DisplayMetrics()
        if (android.os.Build.VERSION.SDK_INT >= android.os.Build.VERSION_CODES.R)
            context.display?.getRealMetrics(displayMetrics)
        else
            @Suppress("DEPRECATION") (context.getSystemService(Context.WINDOW_SERVICE) as WindowManager).defaultDisplay.getMetrics(displayMetrics)

I found some explanations here: https://stackoverflow.com/questions/69150674/android-11-displaymetrics 

but I'm even not sure if this is only a problem related to my phone and OS.

MilosKozak, 2022-02-20:
create PR pls

Andries-Smit, 2022-03-08:
Duplicate of https://github.com/nightscout/AndroidAPS/issues/974

MilosKozak, 2022-05-27:
fixed by https://github.com/nightscout/AndroidAPS/commit/60f5115ce63d91ee2330dfc0dac0eb6fd301b538

closed 2022-05-27 by MilosKozak