nightscout/AndroidAPS #1330: Proposal: remove wizard collators options

Andries-Smit, 2022-02-16:
The Bolus Calculator wizard got a new option since version 3.0 by @Philoul . There are icon indicators on the bottom section that shows what options are used in the calculation. 
[image]
On click of an item it toggles the setting, however, I see some risk with the implementation:
* From a UX/UI perspective, users will not intuitive understand that it is actual checkboxes
* They are close the cancel button and can be clicked accidentally
* When toggled, the setting will be stored, and used in the next calculation too

We risk that the user will accidentally change the parameters and only will figure out after many wrong boluses.

My proposal is to remove the click interaction and only show the settings, so the settings can only be changed when the show checkbox is checked. 

Please let me know what your perspective is?

swissalpine, 2022-02-16:
You're right, you have to know that these are buttons - but that's also the case in other places.
More important fo me: I love the new "buttons"! They are really handy!
However, I had to adjust the display for my small smartphone, so now there is no danger of clicking them by mistake (and I changed so that they no longer disappear when the calculation is shown) :
[image]

Andries-Smit, 2022-02-16:
@swissalpine Your layout is nice, it has a bigger touch area and less change for accidental clicking

Philoul, 2022-02-16:
I think it's a good idea to keep the click in these icons (that's why I made PR #1156 😉) and I identified the risk of too small icons not easy to click when I made this PR so I'm not surprised.
I think it's a good idea to increase the size of Icons on a seperate line (there is enough place on all screen when detailled information is hidden)

I think we can also improve the logic between IOB and COB icons/checkbox (it's another subject but linked because we don't see on Icons if a click is enable or not, checkbox is greyed when disabled...).
- Currently IOB checkbox is checked and disabled when COB is checked (for icons, no action is done if you click on IOB icon while COB icon is "colored"). For safety it's dangerous to have cob checked and iob unchecked.
- Currently when both are unchecked, if you check COB, then IOB is automatically checked and disabled (IOB icon automatically colored and no action available on click)

My proposal would be to always keep both icons (and checkboxes) enabled, check both when COB is checked, uncheck both when IOB is unchecked, and allow IOB checked alone if the user check IOB or uncheck COB...

That way we can always keep an action on all click on icons and keep security (forbid COB checked alone without IOB).

Note for @Andries-Smit in the past this security logic between COB and IOB was not included in Wear, I don't know if it's still the case today (we should add it if missing).

Philoul, 2022-02-16:
We have to update Wear settings (it's in aaps...)
- we have to merge Bolus IOB and Basal IOB to be consistent with wizard
- I think we should implement COB/IOB safety (as you can see in screenshot below we can enable COB without IOB for wizard calculation launched from Wear)
- If we put icons in a seperate line, we can add an optional icon for TT (only visible if a TT is running)...
[image]

MilosKozak, 2022-02-16:
initial idea was make a border around. but I failed to implement it so I ended up with changing alpha

MilosKozak, 2022-02-16:
@swissalpine 's change is nice too.
at the end it cold be possible to remove checkboxes from calculations and use only icons.
there is lot of texts and not much space

Andries-Smit, 2022-02-22:
With the proposed re-layout it would not be necessary to remove the click indicators.
For some users, this might still be an overload of the form. And I might open a PR to  show less information in the treatment popups (carbs, insulin, calculator)

closed 2022-02-22 by Andries-Smit