nightscout/AndroidAPS #1339: Glucose value strike-through and incorrect time for minutes ago

dkmsoft, 2022-02-18:
Has happened a couple of times (Rel 3.0.0.1) that the BG number has a stale value strike-through and number of minutes ago not updated. Screenshot shows last BG to be 203 minutes ago when in fact the BG readings have been coming in every five minutes (as evidenced by correct up to date graph and also xDrip readings are all there up to the current time). It's a display issue.

[image]
.

athinab, 2022-02-20:
I've experienced the same issue, too!

LewEasy, 2022-02-20:
I have had the same thing a couple of times now. 

One minute all is fine then suddenly the Overview Page says last reading was XX(X) minutes ago or similar, all the while xDrip keeps updating and the graph in AAPS is also showing these values. 

Using AAPS 3.0.0.1, xDrip+ nightly from 15.02.2022 with Libre 2 on a Xioami Redmi Note 8 Pro (MIUI Global 12.5.6 / Android 11)

MilosKozak, 2022-02-20:
without logs this issue is useles at this is not common situation

closed 2022-02-20 by MilosKozak