nightscout/AndroidAPS #1347: Feature Request: Extend SMS functionality in automation

thbonk, 2022-02-19:
When sending an SMS from an automation it should be possible to
- select the recipients of the SMS instead of sending the SMS to the entire address book.
- add the current geo location as a Google Maps or Apple Maps link.

Background:
- When having many contacts, not all shall receive the SMS, only the closest persons.
- I have setup an automation such that a SMS is sent, if the blood sugar drops below 54. For the case that I drop out, the recipients of the SMS shall know my geo location to send help, if I don't answer calls, SMS or WhatsApp anymore.

Andries-Smit, 2022-02-19:
I dont think AAPD will send it the whole world, :) Check the SMS communcator plugin, there you can set the phone numbers. Please test it out.

The geo location is legit question. Are there other service like find my device you can use for that? As you might be moving in a car... or zo and a once off GPS location might not me good enough

thbonk, 2022-02-20:
Hi @Andries-Smit 

> I dont think AAPS will send the message to the whole world, :) Check the SMS communcator plugin, there you can set the phone numbers. Please test it out.

Ok, I had the SMS Communicator plugin disabled bea use I thought that it is only used for remote controlling AAPS. You're right, It's possible to configure the mobile phone numbers.

> 
> The geo location is legit question. Are there other service like find my device you can use for that? As you might be moving in a car... or so and a once off GPS location might not me good enough

Hmm, I'm not an Android guy. On iOS, Find My Device is only available for the device owner. On iOS it's also possible to share the location with other people and then they can see your position all the time. But this is a huge privacy issue IMO. I only want to share my location in case of an urgency.

Your remark regarding moving in a car is kind of legit. On the other hand driving a car with an improper blood glucose level is careless. In Germany the driver license is gone, if the police catches you with a BGL < 80. And it might be gone for years, because you have to undergo medical-psychological examinations.

MilosKozak, 2022-02-20:
i don't think this should be resolved the way described here

closed 2022-02-20 by MilosKozak