nightscout/AndroidAPS #1348: NS Client has not temp targets

timocm, 2022-02-20:
Any change in making a temporary target in AAPS 3.0.0.1 is not showing up in NSClient app 3.0.0.1
[image]
[image]

timocm, 2022-02-20:
[Uploading AndroidAPS.log…]()

MilosKozak, 2022-02-20:
to be able look at the issue i need log from both and time of occurence

timocm, 2022-02-21:
temp target in AAPS started at 9.51pm and stopped at 9.57pm

AndroidAPS 3.0.0.1
Build: 3bbb57f33-2022.02.04-00:20
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.6

[AndroidAPS_LOG_1645445063073.log.zip](https://github.com/nightscout/AndroidAPS/files/8109046/AndroidAPS_LOG_1645445063073.log.zip)

------------------
no temp target recieved in NSClient at 9.51pm and when Temp target was stopped 9.57pm

NSClient 3.0.0.1
NSCLIENT
Build: 3bbb57f33-2022.02.04-06:59
Remote: https://github.com/nightscout/AndroidAPS
Flavor: nsclientrelease
Nightscout version: 14.2.6

[AndroidAPS_LOG_1645444790168.log.zip](https://github.com/nightscout/AndroidAPS/files/8109078/AndroidAPS_LOG_1645444790168.log.zip)

timocm, 2022-02-21:
temp target in AAPS started at 9.51pm and stopped at 9.57pm

AndroidAPS 3.0.0.1
Build: 3bbb57f33-2022.02.04-00:20
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.6

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8109133/AndroidAPS.log)

no temp target recieved in NSClient at 9.51pm and thus Temp target was stopped 9.57pm

NSClient 3.0.0.1
NSCLIENT
Build: 3bbb57f33-2022.02.04-06:59
Remote: https://github.com/nightscout/AndroidAPS
Flavor: nsclientrelease
Nightscout version: 14.2.6

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8109141/AndroidAPS.log)

timocm, 2022-02-21:
Temp target started at 11.03pm in AAPS
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8109367/AndroidAPS.log)

timocm, 2022-02-21:
No temp target in NSClient at 11.03pm

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8109396/AndroidAPS.log)

janvitos, 2022-03-01:
I've observed the same issue. Temp targets are not sent from AndroidAPS 3.0.0.1 to NSClient 3.0.0.1.

It seems like it was working when I first installed both apps, but it stopped working at some point.

I've rebooted both phones but to no avail.

closed 2022-03-05 by timocm

janvitos, 2022-03-08:
Hi @timocm, would it be possible to know why you closed this ticket? I am still experiencing this issue.

Did you do anything to solve the problem?

Thanks.

timocm, 2022-03-08:
Started to use Dexcom G6 (vs G5); did another update to AAPS 3.0.0.1 (I had
returned to 2.8.2 while using G5) and then noticed that temp targets were
going through.