nightscout/AndroidAPS #1349: DynamicISF safety limiter bug

robertrub, 2022-02-20:
@MilosKozak If I select the DynamicISF option, I have a safety limiter at 1.5 (mine is 10U) which may (or may not) come from my basal rate (0.5) x max multiplier (3).

[image]

This is a game stopper of course ;) No more SMBs, no more high basals etc.

If I select OpenSMB, the same limiter goes to 22.

[image]

philipgo, 2022-02-20:
Same problem with latest dev and AutoISF here.

bchris21, 2022-02-20:
Same here (3.0.0.1-dev-a) but with different value, maybe because it refers to "child" profile.

Screenshot with SMB last run on 10:56 is with Dynamic ISF

[image]

Screenshot with SMB last run on 11:00 is with OpenAPS SMB plugin.

[image]

Alamo04, 2022-02-20:
Confirm, same here.

elgrupo123, 2022-02-20:
unfortunately also here.  IOB max 4IU

vanelsberg, 2022-02-20:
Confirming,
Duplicate of https://github.com/nightscout/AndroidAPS/issues/1346

swissalpine, 2022-02-20:
Please note: The first commit (https://github.com/swissalpine/AndroidAPS-1/commit/a0c3fad06ea3dad73e3819794e7889e427ccb664) is wrong, the second works for me:
https://github.com/swissalpine/AndroidAPS-1/commit/6a961eb4d8b949ff8564bcd17f0f3b2469e0beda

closed 2022-02-20 by MilosKozak

Alamo04, 2022-02-20:
@MilosKozak @swissalpine 
There are 2 DynamicISF errors in building the App, looks like kotlin errors.
Think they are not importand and self fixed in this version.
[image]