nightscout/AndroidAPS #1356: [AAPS 3.0.0.1] Master password & biometric popup at the same time, on app start

JazzZ1, 2022-02-20:
Time of occurance in log: 20.02.2022, 21:37

Description: I have set all security settings to biometric:
[image]
Still sometimes on app start there are two popups (In front: Biometric; in the back: master password) showing at the same time (I can't make a Screenshot of that, because of Android security reasons). So i first have to do biometrics validation, then I have to put in the master password.

Build: 479bfc410-2022.02.06-18:41

JazzZ1, 2022-02-20:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8104900/AndroidAPS.log)

Andries-Smit, 2022-03-05:
App protection is called on resume, and did not care if it was already asking for authentication.... When the bio authentication is already active, Android cancels it. Therefor falling back to app master password.
It reproduce the issue, change orientation of the app. Or open and close app multiple times without authenticating.

This issue will be fixed in PR #1409 commit cfa64162b0b5c4d6a60916bec55087b2df758a0c

closed 2022-03-14 by MilosKozak