nightscout/AndroidAPS #1359: Riley Link Orange can't connect with dev branch

macarolobe, 2022-02-22:
Impossible to connect the Riley Link Orange with dev branch. Connection is ok with master branch.

macarolobe, 2022-02-23:
It's the RL check for permission before connect code the problem https://github.com/nightscout/AndroidAPS/commit/b1d45833c5570910142e78ba39a8322b4039a9d5

With master branch code no problem to connect RL Orange.

MilosKozak, 2022-02-24:
you should see notification to request permissions
add logs if you can reproduce

macarolobe, 2022-02-25:
No notification to request permission. The state of RileyLink is stuck on initialization. RileyLink Orange FW 3.2 / Samsung S10e with android 12
[AndroidAPS_LOG_1645777569074.log.zip](https://github.com/nightscout/AndroidAPS/files/8139619/AndroidAPS_LOG_1645777569074.log.zip)
[image]
.

macarolobe, 2022-02-26:
It was with android 12 ? I tried to restart aaps and android but nothing change, the RileyLink Orange is stuck to initialization with NO notification to request permissions. I've changed code to connect without request permissions and it's ok.

seattlebrighton, 2022-03-01:
I'm having the same problem with dev branch and Android 12 on pixel 5. Rileylink stuck in initialization with no request popup, even after restarting Bluetooth, phone, Rileylink. Went back to master branch and it connected immediately.

jroth-haj, 2022-03-04:
@MilosKozak 
I have the same problem with Pixel5 (Android 12), MDT and EmaLink / RileyLink.
Also no authorization request after reboot.
AAPS 3.0.0.1 connects in a few seconds.

piecycle, 2022-03-12:
I can confirm this. I was running dev from February 4th without any issues, but when trying to update to the latest dev yesterday, I just couldn't establish a firm connection to my OL. In fact, if I scan for it in the Omnipod Eros settings, it is found immediately and I can also set it as my RileyLink, but the OL doesn't vibrate to indicate a real connection, and indeed the RileyLink Status on the Omnipod Eros screen is stuck at "RileyLink initialization...". I have tried reinstalling several times (not deinstalling the app), clearing the AAPS cache, clearing my bluetooth cache. Absolutely nothing worked. When I downgrade my AAPS to the older dev version from February, it will connect immediately. I have also tried connecting without a currently active Pod (i.e. between pod changes). Didn't help. As the OP above also used Pods, and seemingly also Eros pods, I can imagine that it is related to the Omnipod Eros driver, perhaps.

Running AAPS on a Pixel 3 XL with Android 12. No authorization request after reboot.

Attached my logs (set to: CORE, APS, OHUPLOADER, PUMP, PUMPBTCOMM, PUMPCOMM, PUMPQUEUE, PROFILE - hope that's enough?). Within the last hour I have tried getting the connection to work several times.
The last attempt was at 9:13, here's what I did:

- 9:13: Install latest dev over my (modified) version from Feb 4th
- Wait a few seconds when AAPS tried to connect, without success
- Went to the Eros preferences. My OL was already selected, but I did a new scan and selected my OL again. It did not vibrate and the driver was stuck at RileyLink initialization.
- Waited for about 2 mins as AAPS was trying to connect
- 9:16: Reinstalled my Feb 4th version. Reconnecting to my OL took about 6-7 seconds.

The logs contain traces of the plugins I am developing flagged as 'Tsunami', but the version I installed at 9:13 (and a few times before that) was a clean vanilla dev build without any mods as pulled from the dev branch so my plugins are not connected to this issue (and I also never touched any pump drivers). Let me know if you need more data!
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8237289/AndroidAPS.log)

piecycle, 2022-03-12:
I can also confirm that the code change described by OP works as a workaround. In "RileyLinkBLE.kt" inside the function "connectGattInternal", changing

`if (config.PUMPDRIVERS && Build.VERSION.SDK_INT >= Build.VERSION_CODES.S && ...`
to
`if (config.PUMPDRIVERS && Build.VERSION.SDK_INT > Build.VERSION_CODES.S &&`

allows me to connect to my OL without any issues.

happygirlnyc70, 2022-04-13:
I have been able to connect quickly and reliably to OL using dev-i, build be604bc3b.

jroth-haj, 2022-04-13:
Yes, seams to be solved.

closed 2022-04-13 by MilosKozak