nightscout/AndroidAPS #1360: (3.0 master and dev) basal y-axis cutoff in profile editor- minor

happygirlnyc70, 2022-02-22:
Minor issue
Y axis cut off in profile editor for basal graph
Pixel 4a 5g Android 12

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
[image]

Andries-Smit, 2022-02-22:
@happygirlnyc70 it should be fixed it in https://github.com/nightscout/AndroidAPS/pull/1292
Could you confirm?

happygirlnyc70, 2022-02-22:
@Andries-Smit it is still present in dev-a
[image]

happygirlnyc70, 2022-02-24:
This is a minor cosmetic issue.  I will close this ticket.

closed 2022-02-24 by happygirlnyc70