nightscout/AndroidAPS #1361: If battery dies during bolus delivery you get full bolus anyway but AAPS does record only part of it

PetrMlejnek, 2022-02-22:
CASE:

I hit bolus shortcut for breakfast that should give me 8,7 U.
During bolus delivery I get error message in AAPS that only 1,1 U was delivered.
I looked at the pump display that shows me the battery needs to be replaced.
I was sure that I get more than 1,1 U of insulin because insulin delivery took a long time (definitely longer than 1,1 U).
I replaced the battery.
I synced pump with AAPS by hitting Bluetooth logo in AAPS.
But AAPS still shows that only 1,1 U was delivered.
So I looked to pump bolus history and there was a record of 8,7 U delivery.
So pump knows that it delivered full bolus, but AAPS doesn't.

So the DANGEROUS problem is that if battery dies during bolus delivery you get full bolus anyway but AAPS does record only part of it. If you did not notice it, you can overdose and end up in hypo.

--------------
PUMP: Dana R v2, Model: 03, Protocol: 02, Code: B1
APP: Android APS 2.8.2

[image]
[image]
[image]

MilosKozak, 2022-02-22:
do you have log files?

PetrMlejnek, 2022-02-22:
It happened this morning between 8 to 9 am.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8119333/AndroidAPS.log)
[AndroidAPS._2022-02-22_00-00-03_.2.zip](https://github.com/nightscout/AndroidAPS/files/8119334/AndroidAPS._2022-02-22_00-00-03_.2.zip)
[AndroidAPS._2022-02-22_00-00-03_.1.zip](https://github.com/nightscout/AndroidAPS/files/8119335/AndroidAPS._2022-02-22_00-00-03_.1.zip)
[AndroidAPS._2022-02-22_00-00-03_.0.zip](https://github.com/nightscout/AndroidAPS/files/8119336/AndroidAPS._2022-02-22_00-00-03_.0.zip)

PetrMlejnek, 2022-02-22:
It happened this morning at 8:21.
[AndroidAPS._2022-02-22_00-00-03_.1.txt](https://github.com/nightscout/AndroidAPS/files/8119353/AndroidAPS._2022-02-22_00-00-03_.1.txt)

MilosKozak, 2022-02-25:
now i see it's 2.8
logging was not so good to be able find anything.
and you are not using v2 driver anyway. main R driver doesn't use history and any problem with connectivity leads to wrong data almost everytime

closed 2022-02-25 by MilosKozak

PetrMlejnek, 2022-03-03:
Omg, I don't know how that happened. Thank you!